arkypita/LaserGRBL #451: Experimental SVG support

arkypita, 2018-09-30:
Write here about SVG support

StuartB4, 2018-10-01:
Hi.
Just a bit of info.
Just installed v2.9.0 and made a quick test .SVG file.
This was the result:

I left the settings at the default ones, loaded the .svg file in which was made in Inkscape.
I noticed the M4 command to switch the laser on, like in the picture below. 
But after generating the code and saving it, I looked at the code and saw it had generated M3 instead for the laser on.

[image]

I also noticed in the code there is no return to home position code generated.
So when it had finished burning, the laser stopped at the (bottom right) end of the last line.
Also there was an off-set of X1 and Y2 mm's added.

[image]

Generated code.

[image]

This is the finished test burn.
It burnt the shapes in this order:
1: Square.
2: Spiral from center.
3: Circle.
4: Left Star.
5: Right Star. 

[image]

StuartB4, 2018-10-02:
Hi Arkypita.
Not sure about the off-set. 
All I did was to draw the shapes where they are and save the .svg,
I did'nt take any notice of sizes or position, just made sure they were all inside the 40 x 40 mm square. 
I just had a look at the .svg file and marked the X1 Y2 position.
So although it was'nt added intentionally it was ok.

[image]

> It is an issue or it is ok? I think that the code is generated as found inside xms-svg code, with few control on it by my side.

Certainly not an issue for me, just an observation.
As long as it burns correctly I don't think it matters that much.

StuartB4, 2018-10-02:
Did you remove Line To Line Tracing from version 2.9.0 ?
It appears to be missing.

arkypita, 2018-10-03:
> Did you remove Line To Line Tracing from version 2.9.0 ?
> It appears to be missing.

question removed?
no changes to raster import in 2.9
configure with "pwm support" to see line2line option

sanyats, 2018-10-03:
Здравствуйте. Все хорошо, открывает. Возможно сделать маштабирование векторов. Спасибо

sanyats, 2018-10-03:
в режиме SVG не позиционирует в центр. 
[image]

arkypita, 2018-10-03:
translated: Hello. All is well, opens. It is possible to do vector scaling. Thank you

Yes, as future option I will add scaling.
For now I suggest to use inkscape to prepare the image at the required size.

Luborson, 2018-10-10:
Hi, GOOD Job !
i have some question:
1. do you plan for filling of vectors ? (power depend for colour ?)
2. numbering of generated lines Gcode (ON/OFF)
3. every new append file have first move x0 y0, can by switchable to ON/OFF
4. same like 4, but for vectorized and filled picture, outline start again from x0 y0
5. now not work "start from specific line over green START arrow", startet again from first line, not from specified line, but ..File/Send From Possition is work well

thanks

MarceloFreitas028, 2018-10-15:
I also test the corel X7 SVG exportation and works well. I think that if you can develop the vectors fill it can be very useful. i export the same file filled and unfilled an this is the difference in the files.

unfilled:
[image]

Filled:
[image]

arkypita, 2018-10-15:
> 1. do you plan for filling of vectors ? (power depend for colour ?)

I can't exclude. Maybe in the future but not for now.
For now i suggest to use filling functions from your favourite SVG editor: https://wiki.evilmadscientist.com/Creating_filled_regions

hayttle, 2018-10-17:
I can´t import SVG file.
[image]

hayttle, 2018-10-17:
The same with new version.
[image]

arkypita, 2018-10-18:
Can you submit your svg?

hayttle, 2018-10-18:
SVG is not supported here. :(

arkypita, 2018-10-18:
Zip and upload

hayttle, 2018-10-18:
I found the error.
The file was generated from Vectric Aspire with character coded accent and the software could not deal with this.

arkypita, 2018-10-19:
I was sure that the problem was in the file, but if you can share it (the one with issues) i can try to make the import procedure more strong and error tollerant

hayttle, 2018-10-19:
Of course.

Another feature that would be great is can import dxf file.

Here it is.
[error_file.zip](https://github.com/arkypita/LaserGRBL/files/2497462/error_file.zip)

Thanks and congratulations for the good job.

arkypita, 2018-10-21:
After a little search i found that [xml specification](https://www.w3.org/TR/xml/#charsets) define that only some character ranges are allowed in xml (svg is a form of xml file). So your file is not a valid xml file.

I have tried to strip-out any character out of valid ranges prior to parse file content and now I was able to correctly read your error_file.svg

Future version of LaserGRBL will be more strong!

hayttle, 2018-10-21:
I'm happy to contribute to the project.
Nice job.

Do not forget to allow direct import of the dxf file, without having to convert to svg.

;)

StuartB4, 2018-11-03:
Hi Arkypita.
Just been trying SVG files again.
I noticed when it generates the GCode it writes M5 twice at the top of the code.
Not a problem, just wondered why it does it.
Start Of Code:

G90
M5 
M5
G0X39.6Y19.81
M4 S255 
G1X39.58Y19.86 F250
X39.52Y19.92
X39.43Y19.97

The other thing I noticed is when changing the setting from M4 to M3 just before generating the GCode 
It still writes M4 in the code.
Again, not a problem as it can be changed/edited in the saved GCode file if needed.

hk89, 2018-11-29:
Hi @arkypita 

Just been using the awesome SVG feature. I do have some issue where after an SVG had finished engraving, the machine will go over the same paths several times with the fast speed and laser turned off before going into engraving the second path, third path and so on. I mean after every path, it does that before going to the next.

hk89, 2018-12-04:
Bump. Any ideas?

StuartB4, 2018-12-04:
Strange problem.
I just downloaded your file, loaded it in to Inkscape and shrunk it down to fit my machine because it's only 40 x 40 mm's, saved it and loaded it in to LaserGRBL and it burned perfectlly.

For some reason though when loading it in it also showed the original size path in the window
shown in the image below.

After it burned ok it then satrted to go over the original size path but without the laser coming on. 

[image]

As good as it's can be at this size.

[image]

StuartB4, 2018-12-04:
Just tried it again by cutting the object from the original document and pasting it in to a new 40 x 40 mm document. Shrunk it again to fit the document and burned it again.
This time it did it perfectly and just once.

May be worth a try to just select the object, cut it and paste it in to a new document the same size as the original.

[image]

[image]

StuartB4, 2018-12-05:
Hi hk89.

Had a look at the file with Notepad++.
There are 3 seperate large blocks of characters in the file.
Looks like way to much for what it's actually got to do.
I removed the last 2 blocks and re-saved the file
Could you just try this file Just on a piece of scrap when you have a moment.
It worked on my machine perfectly and only did it once, as it's supposed to.

[12.zip](https://github.com/arkypita/LaserGRBL/files/2648712/12.zip)

garthanstey, 2018-12-24:
I've just got into svg files from inkscape and tried loading them into lasergrbl . The images are great but I have the problem of getting them the exact size I need. Also I have built a jig at 100mm x and y offsets so that I can rapidly repeat burns. The option to resize and offset is not present for svg and i understand you are working on this. I'd love to know how things are going.

RudiZc, 2019-01-11:
Hi.
I noticed just a small issue using svg conversion: I'm using illustrator and if I use millimiters as unit, when I import my file the scale change and become bigger. I found that that scale matches with the dimensions of the image if I use "points" as unit in illustrator. So now, to have the right dimensions importing an svg file, i change my measure unit in illustrator from millimiters to points and consider that as millimiters.

chrisgrul, 2019-01-25:
> Hi.
> I noticed just a small issue using svg conversion: I'm using illustrator and if I use millimiters as unit, when I import my file the scale change and become bigger.

I'm having the same problem. It seems when using Illustrator, LaserGBRL considers mm to be pt and divides by about 2.83465. I had been dividing my measurements by this value but I like Rudi's suggestion as it is easier.

svenhb, 2019-01-25:
Yes, usually scaling is 96 dpi (inkscape) but Adobe Illustrator uses 72 dpi.

arkypita, 2019-01-28:
Hi all, and thanks for feedback about SVG support.

I'm aware of these problems, unfortunately fixing is not easy because the code I use to load SVG was not written by me and is a bit complicated to modify-understand.

Maybe in the future I will be able to solve them but at the moment I do not have much time to do it.

What I can suggest is to use inkscape to convert them from the original svg format (whatever the program that generated them) to inkscape svg. Open them in inkscape, set the document property to mm, adjust size and position of your job in page and save again.

What I've seen is that the SVG import routine work pretty well when the files are produced like that.

StuartB4, 2019-05-17:
This is a bit strange. Since updating to version 3.0.14 LaserGRBL refuses to load .svg files, no matter which software I use to save them in .svg format, even Inkscape.
I select the image then click the Create button and nothing happens, just a blank/empty window.

Any ideas why would be appreciated.

arkypita, 2019-05-17:
Very strange! No changes to svg import code.
Tried now, it work. Please test with this file:

[disegno.zip](https://github.com/arkypita/LaserGRBL/files/3192010/disegno.zip)

StuartB4, 2019-05-17:
Thanks Arkypita.
Thats fine, star and spiral.
Seems to be when I import .png files in to Inkscape and save them as .svg it does'nt work.
I must be missing something.

arkypita, 2019-05-17:
I think you should vectorize them using inkscape vectorize bitmap.
Raster image embedded in svg are not supported, neither in latest or older version.

StuartB4, 2019-05-17:
Ok, Thanks very much.

lbogaardt, 2019-09-09:
Hi Arkypita,

Thanks for your work on LaserGRBL.

I recently switched from EleksCAM to your software. I am hoping to adjust SVGs after I import them. In particular, I would like to be able to:
- Translate
- Scale
- Rotate

I understand you currently do not have a lot of time to work on this. Could you give  us an update on the following two questions:
- Do you think you'll be able to add such SVG features soon?
- Do you still recommend using InkScape to change the original SVG before importing it into LaserGRBL? Or are there other easy work-arounds?

Rybec, 2019-12-25:
For those with scaling issues using Illustrator:  Please contact Adobe about this problem.  Inkscape uses 96 dpi, because 96 dpi is _the_ official standard for SVG.  (This standard is inherited from XML, which inherits it from CSS.  SVG uses CSS attributes for position, size, and such, and 96 pixels is defined as 1 inch in CSS, thus in SVG, 1 inch is canonically 96 pixels.)  Adobe uses 72 dpi in Illustrator, which makes their files non-compliant with the SVG/XML/CSS standards.  (One might reasonably say they are not even real SVG files, even though SVG parsers can load them mostly correctly.)  No other software should have to go out of its way to support non-compliant software.  This means the appropriate resolution to this problem is to get Adobe to make Illustrator compliant with the SVG standard, _not_ to request other software, like this, to jump through extra hoops because some company that is too big to care about its _paying_ customers won't comply with the official standards.

My solution is to use standards compliant software, like Inkscape, when I am doing any kind of work that is likely to be translated into real world objects.  And to be perfectly honest, I find Inkscape significantly easier to use than Illustrator.

arkypita, 2020-04-08:
Hi @Rybec I totally agree with you!
SVG is an open standard or, as I call them, a "non standard standard".

Apart from a series of basic rules that establish the formal validity of the file, the SVG format is subject to different interpretations and implementations that vary from program to program, which have the freedom (since it is an open format) to add their own keywords and specific features.

Also to reply to @lbogaardt my goal is to be as compatible as possible with inkscape svg, because inkscape is free, easy, and open.

My SVG import routine comes from [GRBL-Plotter](https://github.com/svenhb/GRBL-Plotter) project, which also has the possibility of importing .dxf and modifying the G-code. Both possibilities that I would like to import into LaserGRBL as soon as possible (which could mean months!). Until then my advice is to use inkscape to produce and modify the vectors (including the dxf which are loadable in inkscape and convertible to svg) in order to get the exact job you want to mark, including size and offset.

It is better to use inkscape v0.91 because it was reported that there are issue with SVG format generated by inkscape 0.92 when SVG generated by inkscape 0.91 is always good for LaserGRBL.
I have not yet managed to get my head on this problem but I plan to do it in the next few days.

arkypita, 2020-04-08:
Looking at the code of the project from which I took the SVG library I see that in the comments it says:

```
/*  GCodeFromSVG.cs a static class to convert SVG data into G-Code 
    Not implemented: 
        Basic-shapes: Text, Image
        Transform: rotation with offset, skewX, skewY

    GCode will be written to gcodeString[gcodeStringIndex] where gcodeStringIndex corresponds with color of element to draw
*/
```

This is exactly the cause of the problem reported [here](https://github.com/arkypita/LaserGRBL/issues/907#issuecomment-598447426)  by @Ziggy2013

Looking at the code of file GCodeFromSVG.cs line 319 we can see that the code use only the param "angle" and ignore the rotation point.

```
if ((transform != null) && (transform.IndexOf("rotate") >= 0))
{
	var coord = getTextBetween(transform, "rotate(", ")");
	var split = coord.Split(',');
	if (coord.IndexOf(',') < 0)
		split = coord.Split(' ');
	float angle = floatParse(split[0]) * (float)Math.PI / 180;
	tmp.M11 = Math.Cos(angle); tmp.M12 = Math.Sin(angle);
	tmp.M21 = -Math.Sin(angle); tmp.M22 = Math.Cos(angle);

	if (svgComments) gcodeString.Append(string.Format("( SVG-Rotate {0} )\r\n", angle));
}
```

Now I'm looking for a trigonometric solution (I'm not very good with this) to implement rotation around a point other than 0.0

arkypita, 2020-04-08:
It would seem to be easy, just replace the assignment of M11, M12, M21, M22 of a transformation matrix with the call to the RotateAt function.

```
if ((transform != null) && (transform.IndexOf("rotate") >= 0))
{
	var coord = getTextBetween(transform, "rotate(", ")");
	var split = coord.Split(',');
	if (coord.IndexOf(',') < 0)
		split = coord.Split(' ');
	float angle = floatParse(split[0]);// * (float)Math.PI / 180;
	float px = split.Length == 3 ? floatParse(split[1]) : 0.0f;
	float py = split.Length == 3 ? floatParse(split[2]) : 0.0f;

	//tmp.OffsetX = px;
	//tmp.OffsetY = py;
	//tmp.M11 = Math.Cos(angle); tmp.M12 = Math.Sin(angle);
	//tmp.M21 = -Math.Sin(angle); tmp.M22 = Math.Cos(angle);
	tmp.RotateAt(angle, px, py);
	if (svgComments) gcodeString.Append(string.Format("( SVG-Rotate {0} )\r\n", angle));
}
```

Tested with file pent with vines [here](https://github.com/arkypita/LaserGRBL/issues/872) and now is imported correctly. I am going to do more test.

Ziggy2013, 2020-04-08:
Thanks arkypita for following up on this issue.

An explanation of how to handle rotation about a point other than the origin is here on youtube-

https://www.google.com.au/search?source=hp&ei=16iNXuquOOGW4-EPw_2vyAg&q=rotation+about+a+point+not+the+origin+formula&oq=rotation+about+a+point&gs_lcp=CgZwc3ktYWIQARgIMgIIADICCAAyAggAMgIIADICCAAyAggAMgIIADICCAAyAggAMgIIADoFCAAQgwFKNwgXEjMxNTJnMTY4ZzE2MGcyNDRnMTU4ZzE1OWcxNjRnMTUxZzE2MGcxNjFnMTU4ZzE2M2cyMTRKHQgYEhkxZzFnMWcxZzFnMWcxZzFnMWcxZzVnNWczUMkiWJp4YIiwAWgAcAB4AIABtAKIAfockgEIMC4yMS4xLjGYAQCgAQGqAQdnd3Mtd2l6sAEA&sclient=psy-ab#kpvalbx=_7qiNXqDBM7aO4-EP3pOisAQ40

arkypita, 2020-04-08:
Release v3.1.4 (available as pre release for test) should fix the issue of rotation about point.
https://github.com/arkypita/LaserGRBL/releases/tag/v3.1.4

Let me know

Ziggy2013, 2020-04-08:
Hi arkypita

I think the rotation about a non origin point is now working.

However it appears there is still an issue with how rotate is handled.

I have attached three test files;

1) an svg file "test_stencil_brd.svg"  which been generated out of a PCB
program. This file renders correctly in Inkscape 0.92.3 It
    does not render correctly in LaserGRBL. You will see the difference in
rendering affects just one part of the image.

2) The same svg file "test_stencil_brd.svg" has been openned in Inkscape
0.92.3 then simply saved.  This saved file
     "test_stencil_brd_saved.svg" renders correctly in Inkscape 0.92.3 but
not in LaserGRBL.  If you look at the xml contents
      of file 1 and 2 they are very different. However I think the reason
LaserGRBL does not render correctly is the "rotate(-270.000000)"
      statement in both files "test_stencil_brd.svg" and
"test_stencil_brd_saved.svg" is not handled correctly.

3) The third file "test_stencil_brd_ungroup.svg"  is the result of just
Inkscape "ungrouping" the content of file 1. Both Inkscape
     and LaserGRBL now render this file correctly. If you look at the xml
contents of this file you will see that Inkscape uses
     rotation about a non origin point. This translation type now appears
to work correctly in LaserGRBL 3.1.4.

I hope this information and test files are helpful.  All of these three
files should render the same in Inkscape 0.92.3 and
LaserGRBL. They should also render the same in earlier versions of Inkscape
but I haven't been able to check that.

I can see the svg import code is very complex. If you need further testing
please let me know.  I appreciate the massive effort you
have put into LaserGRBL. It's a great app - thanks.

Regards

Ian

arkypita, 2020-04-08:
Mail attachment does not work in github. Can you zip and load via github web page, or send them to info@lasergrbl.com?

Ziggy2013, 2020-04-08:
Hi,

Just sent the zip file to  info@lasergrbl.com

regards

Ian

arkypita, 2020-04-09:
Thanks @Ziggy2013 
In addition to the lack of "rotate at point" the problem seems to be more the ability to manage multiple transform operations (rotate translate) in the same transform attribute. In particular, the order in which these transformations are indicated because the order in which they are performed is important (while the library I used ignored this detail).

Now it should be fine with your files and, optimistically speaking, with every file.
Let me know

https://github.com/arkypita/LaserGRBL/releases/tag/v3.1.5

Ziggy2013, 2020-04-09:
Hi arkypita,

v3.1.5 correctly renders all three test files - it's perfect - thanks.

I agree the root cause issue is more about the order of thransformations.
There is some confusion about how the W3C specification defines
the order in which transformations should be done - the correct order is
from right to left but the specification does not make it easy to
understand this.

There is an interesting discussion about the transformation order is here

https://stackoverflow.com/questions/18582935/the-applying-order-of-svg-transforms

regards and thanks for fixing the problem

Ian

arkypita, 2020-09-24:
Memo: viewbox issue https://github.com/arkypita/LaserGRBL/issues/1126

lojzik, 2020-12-06:
Would it be possible to generate PWM and multiple passes count on elements from some element attribute (opacity, color, stroke-width, etc.)?

arkypita, 2020-12-08:
Already in #64 development roadmap

mtk11, 2021-02-14:
> > 1. do you plan for filling of vectors ? (power depend for colour ?)
> 
> I can't exclude. Maybe in the future but not for now.
> For now i suggest to use filling functions from your favourite SVG editor: https://wiki.evilmadscientist.com/Creating_filled_regions

How about when the outline and filling requires different feedrate and laser intensity ?

ashkan4818, 2021-02-21:
Draw your design using CurlDra software, then set the design scale to 10% and convert it to svg. Everything will be fine.

maxim-kukushkin, 2021-03-13:
I've just tried loading an SVG file with a single circle and LaserGRBL v4.2.1 seems to interpret it incorrectly (SVG and LaserGRBL-generated NC files attached in the archive)

It's shown properly in the preview, but the GCode (G3 command) is malformed. It also fails to show the frame doing the following commands (see the screenshot): it simply move to a single point (circle center) and back
[image]

[circle.zip](https://github.com/arkypita/LaserGRBL/files/6135225/circle.zip)

Is it a known issue or something settings are not correct

arkypita, 2021-03-13:
If I open your .svg with LaserGRBL v4.2.1 the generated gcode I have is correct.
The preview show that the boundary is computed correctly.

[image]

The exported output is:

```
G90 (use absolute coordinates)
G92 X0 Y0
M3 S0
S0 
G1X156.86Y15.47F1000
S1000 
G3X156.86Y15.47I-5.88J0F1000
S0 
M5 S0
G0 X0 Y0 (move back to origin)

```

maxim-kukushkin, 2021-03-13:
Hmm... but I downloaded it from here: https://github.com/arkypita/LaserGRBL/releases and used the file that I provided. What could influence the processing of this file? Are there any runtime libraries that LaserGRBL relies on for generating paths?

arkypita, 2021-03-13:
Download location is correct.

> What could influence the processing of this file?

Nothing in my mind.
I don't remember it use data from configuration or grbl settings.

> Are there any runtime libraries that LaserGRBL relies on for generating paths?

Maybe some issue with OS localization (internationalization, language settings, decimal point character)?

arkypita, 2021-03-22:
I am sorry but I have no time now to follow the project and SVG issues.

> It makes it possible to LaserGRBL on Linux as well, which is quite important as there's currently shortage of free (and especially open source) software for lasers.

FYI It was reported that LaserGRBL run good on Linux with wine 5.0 and wine mono 4.9.4 (wineprefix 32-bit) and windowsdll gdiplus.

alastaira, 2021-05-15:
If the SVG code is rewritten at some point in the future, one really useful feature would be the option to process any interior paths contained within a shape before cutting the exterior outline of the shape itself. 

Currently, I think LaserGRBL prepares the gcode by processing coordinates in the order in which they are presented in the SVG, which is normally exterior path first, followed by any interior "holes", but this is a problem for cutting interior details in a shape that might shift slightly after cutting (or, worse, fall out of the bed completely!)

My workaround for this is to separate any interior holes onto a separate layer in Inkscape, and process that first. But it would be nice if that extra step could be avoided.

Rybec, 2021-05-15:
It has been a while since I've done anything with this, but I _think_ level affects order.  If you are willing to experiment, try this: Once your design is finished, select all of the elements you want to cut first, and press the "End" button on your keyboard.  If you are using a Mac or a crippled laptop keyboard, you can use the menus instead: Object -> Lower to Bottom.  At least in SVG 1, rendering order is based on element order, so moving to the bottom, making an object render earlier, means putting it earlier in the file.

Note that to select inner edges and manipulate them independently, you may have to convert your objects into paths before changing their render order.

(Also note that this is not about _layers_.  Objects have an order within a layer.  You can use Page Up and Page Down to move one order level at a time, or Home and End to bring them all the way to the top or the bottom.  I believe that if you have multiple elements selected when you do this, they will keep their level relative to each other, but will be moved relative to everything else.)

Anyhow, _if_ this works, it would be better than automatically doing that.  I've had a few cases of my own, where I needed the outside cut to go first.  I don't recall how I solved this, but I suspect I did what you did, and did it as separate cutting jobs.

If you try this, please report back here!  I am also curious if this works the way I think it does, but I can't use my laser cutter to actually test it right now.

[6 more replies]