arkypita/LaserGRBL #455: increase Laserpower

CLEred, 2018-10-04:
Hello my friends

Everything works fine except the power of the laser isn't strong enough. I have connected a 500mw  blue purple laser module with TTL control. I searched in the Grbl configuration and I couldn't find anything. I also couldn't find $30, $31 and $32. I'm using the 2.8.50 version.

It would great if somebody could help me out.
Thank you very much in advanced
CLEred

StuartB4, 2018-10-04:
You need to upgrade GRBL to Version 1.1.
That includes $30, 32 and 32
[Get It Here.](https://github.com/gnea/grbl/releases)

StuartB4, 2018-10-04:
When you have done that, do this. :)
Start LaserGRBL and click Connect, make sure you have the correct Com Port selected.
Then go to the Settings on the GRBL menu and make sure PWM is Ticked.

[image]

Then goto GRBL Configuration settings and set $30, 31 and 32 the same as these.
Just click on the number you want to change to highlight it, then click it again and it becomes editable.
When you have changed all 3 settings click on the Write button at the bottom to write it to the Arduino.

[image]

CLEred, 2018-10-05:
Thank you very much for the quick response. I did everything you told me to do. And now I'm able to see $30, $31 and $32. 
But for some reasons the Laser isn't turning on and off. The Laser is on all the time and engraves everything. The Laser is connected to the Arduino pin 11. I'm also using different power supplies for the steppers and the laser. When I use only one power supplie the laser turns on and off but in this case it has not enough power to engrave.
What could be the problem?

StuartB4, 2018-10-05:
What laser driver board do you have?
Do you have a Ground connection from it to the Arduino Ground?

brakthehun, 2018-10-06:
If you're upgrading from GRBL 0.9 to GRBL1.1,, the pinouts are different. The laser might need to moved to the Z axis limit pin.

StuartB4, 2018-10-06:
No brakthelun.
The Z axiz limit pin is now pin 12 on version 1.1.
The only time you would need to connect it to the Z limit pin using version 1.1 is if you were using an old version 3 of the Protoneer CNC shield.

brakthehun, 2018-10-25:
Yeah,, that's what I was talking about - a shield.

WalterHynson, 2019-01-07:
GRBL1.1 works perfect ...the only thing that is needed is a photo pinout with arduino nano...if you could add that to this page please.

StuartB4, 2019-01-07:
This is a comparison between Arduino UNO and Nano.
https://i2.wp.com/www.likecircuit.com/wp-content/uploads/2017/10/Comparison-pinout-Arduino-Nano-and-Arduono-Uno.png
Pin D11 on the Nano is the same as Pin 11 on the UNO.

WalterHynson, 2019-01-07:
Your post has a lot of other useful infomation as to the correct setting of
the switches in the settings menu which apply not only to uno bur also to
nano that is why I requested the additional info