arkypita/LaserGRBL #456: how to get speed more when engraving

cahyadi99, 2018-10-09:
Hello guys, im new here, i was I have done a lot of ways, but when I carve a lineart, it takes a very long time, at the earliest 1 hour

im using eleksmaker A3

StuartB4, 2018-10-09:
It depends on the amount of lines in the image, the size of the image, the speed the engraving is set to in LaserGRBL, the speed settings in GRBL.
Your machine will only go as fast as the GRBL settings will allow.
$110, $111 and $112 are maximum mm's per minute in GRBL settings.
Start LaserGRBL and make sure you are connected.
To change the settings click on the GRBL menu top left.
[image]

Then on GRBL Configuration.
You should have this:

[image]

Change the numbers I have circled and then click the Write button to save them to GRBL.
What you change them to depends on the maximum speed your machine can physically move at
without loosing steps / slipping etc.

Dpsspower, 2018-10-11:
Eleksmaker A3 cannot go much faster than 6000.
Also the higher the speed, the longer is your acceleration/deceleration ramp.
The third point is, that at high speed and raster image, the small Arduino CPU is not able to interpret all the G-code commands. RS232 is limited to 115KBit/s which gives about a maximum of 1000 G-lines per second.
If you have 1000 pixels to engrave, this will take about 1 second to transmit via RS232.
1000 pixel at 10 line/mm is 100mm length. 100mm /s = F6000 mm/Min.
This is the absolute possible maximum for this setup. To be safe, maybe half is ok.
I engrave plywood at maximum F3000 and other parts maybe just F1000.
If going fast at raster engraving, you need lead-in and lead-out areas at the sides of your pictures to let the laser head accelerate. Otherwise you get different strength of the line.
An engaving at A4 can take 3 hours or more, (2000 lines at 6 seconds/line F3000)

BTW: LaserGRBL does not care about the speed you use.

cahyadi99, 2018-10-12:
> 
> 
> It depends on the amount of lines in the image, the size of the image, filling between lines, the speed the engraving is set to in LaserGRBL, and your machine speed settings in GRBL.
> Your machine will only go as fast as the GRBL settings will allow.
> $110, $111 and $112 are maximum mm's per minute in GRBL settings.
> Start LaserGRBL and make sure you are connected.
> To change the settings click on the GRBL menu top left.
> [image]
> 
> Then on GRBL Configuration.
> You should have this:
> 
> [image]
> 
> Change the numbers I have circled and then click the Write button to save them to GRBL.
> You probably don't need to change the $112 settings as it's the Z Axis for laser / router spindle up and down.
> What you change them to depends on the maximum speed your machine can physically move at
> without loosing steps / slipping etc.

This setting for new version or old version? i use old version is normal, but when i download new version, speed is down not like old version

cahyadi99, 2018-10-12:
> Eleksmaker A3 cannot go much faster than 6000.
> Also the higher the speed, the longer is your acceleration/deceleration ramp.
> The third point is, that at high speed and raster image, the small [...]

you know best software for elekslaser A3 ????

StuartB4, 2018-10-12:
> This setting for new version or old version? i use old version is normal, but when i download new version, speed is down not like old version
These are for latest GRBL version 1.1 

You could try T2Laser.
It's not free but you can download the trial version to test it.
[image]

PC Hardware License (purchased from trial software): $39.95
PC Hardware License (purchased from website): $44.95

Dpsspower, 2018-10-12:
> you know best software for elekslaser A3 ????

Well, THIS software here is more than enough for Elekslaser !
I tried many others and most of them are garbage.

StuartB4, 2018-10-12:
I agree Dpsspower. LaserGRBL is excellent compared to the paid for versions of any other.