arkypita/LaserGRBL #459: Raster image conversion possible pitfall

Dpsspower, 2018-10-11:
Hi there,
I do a lot of raster image engraving.
Sometimes I cut out an engraved image by a separate vector cut. But here the problem started.
I recognized that LaserGRBL calculates (or suggests) the second size of a picture when you type in the first size.
For example Bitmap 802 x 801 pixel. 20 lines/mm. Input B = 40mm. 
The software automatically sets H = 39mm, which is false. It is an error of 2,5% in one dimension.
The engraving will be distorted by 2,5% in one direction.
801 / 20 = 40,05mm and 802 / 20 = 40,1mm.
There is no logical reason to round 801 or 802 / 20 down to 39. 
I guess there is a huge rounding or calculation mistake somewhere in the software.
When you type in H = 40 manually, then the software suggests B = 40mm which is correct.
I saw this problem, when I tried to cut out engraving by vector shapes. it NEVER fits!
So if your engraving for example is a 40mm x 40mm circle and you try to cut this by a 40mm diameter circle, it will never fit.
I am wondering why the software just allows picture size in full millimeters. One decimal place would be great. If your source image for example gives a LOGICAL size of 50 x 50,8 mm, you will get a large calculation error. You must be shure that your source image / number of lines per mm results in full integer value in X AND in Y.
When converting pictures which result in non-integer values of pixel / size, you may get into trouble.

gisep, 2018-10-11:
Hi,
i think laserGRBL calculates the decimals even if it doesn't show them.
Maybe the problem is in your $100 $101 travel resolution? I bet you can solve it from there.

Dpsspower, 2018-10-11:
No.
Point1:
Why should the software round 801 / 20 down to 39 (even internal 39.9999 or whatever)?
801 / 20 = 40,05, so must show at least 40 (or internal 40,xxxxx whatever).
Point2:
I can measure the size with 40 x 39mm at the working piece.
If I manually set the 39 suggestion to 40 then the result is correct with 40 x 40mm.
This has nothing to do with mechanical travel or setting in the controller.

Again for quick understanding:
Take a bitmap B802 x H801 pixel and load it with 20 lines/mm resolution.
Type "40" at the B-size and software will tell you 39 at H-size.
When engraved, the result is exactly B40mm x H39mm.
Correct 39 in H-size to 40 manually and the result will be fine.
Dimensions are also displayed in the preview of the software, even the wrong "39".

Naturally I can solve it by manually correct the value.
I just want to tell people about this pitfall if they trust the software keeping the right aspect ratio of the picture when typing just one dimension. It will not!

arkypita, 2018-10-13:
try update :-)

closed 2018-10-13 by arkypita