arkypita/LaserGRBL #461: Crash when homing or aborting job

ejs-ejs, 2018-10-13:
Hi All,

maybe someone else can confirm these issue: 
1. the LaserGRBL (latest one) is crashing into home switches on the homing command.
2. it will also crash into home switches after a job was cancelled. It will never do this after the job is finished. "G0 X0 Y0" will return to the starting position as expected.

ejs-ejs, 2018-10-13:
For the homing cycle, LaserGRBL issues '$J=G90X0Y0Z0F1954', at least for the emulator. The proper commad is '$H', at least for GRBL.
For the 'Abort', the LaserGRBL issued 'G28', while the proper command would be 'G0 X0 Y0' or none at all.

arkypita, 2018-10-13:
You are wrong

arkypita, 2018-10-13:
[image]

arkypita, 2018-10-13:
jogging interface is used for jogging
homing button is used for homing

different button for different function

arkypita, 2018-10-13:
GCode G28 is documented as "Move to origin" and it is the only command that "go to origin" independently to the absolute/relative movement mode currently active. G0 X0 Y0 depends to G90/G91

https://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29

arkypita, 2018-10-13:
The needs on abort is to go to the origin, but you must keep in mind that you can have file that use absolute G90 or relative G91 modes as well, and you don't know what mode is currently active, so G0X0Y0 is not a valid command in some case. 

G28 should be, because it [is documented as "move to origin"](https://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29) 

However G28 behaviour is hardware dependent, and it is not a safe way to go to origin as well. Anyone has a better solution is welcome

arkypita, 2018-10-13:
> or none at all.

Maybe this could be the better choice

ejs-ejs, 2018-10-14:
> 
> 
> > or none at all.
> 
> Maybe this could be the better choice

So you ca also command as 'G90X0Y0', when the job is aborted?

arkypita, 2018-10-21:
next version will come without G28 when abort
however i think you have some misconfiguration issue if going to 0,0 will crash your head to limits

arkypita, 2018-10-21:
> Is it possible to add "G-code setup block" and "G-code exit block" to the program? In most cases i swap spindle to the laser and for this i need to reconfigure the machine - enable 'Laser mode', maybe disable Z homing. And i need the opposite when the software exits. Now all these actions must be made manually.

You can use custom buttons for that

closed 2018-10-21 by arkypita