arkypita/LaserGRBL #462: SVG negative origin values

JakeBrownPDX, 2018-10-16:
Hey Arkypita, I'm excited to kick the tires on the new SVG feature.  I tried throwing some map data at it and I'm getting some negative values for x and y.  Any ideas?  

[image]

StuartB4, 2018-10-26:
Hi Jake.
Just wondered if you found out why it was doing it.

JakeBrownPDX, 2018-10-28:
No idea @StuartB4 .  Issue persists in version 3.  However, I see the new version 3 has centerline tracing which will actually give me what I need.  Nice work @arkypita , I plan on making a donation.

StuartB4, 2018-10-28:
Ok Jake, thanks for the reply.

arkypita, 2018-11-19:
hi @JakeBrownPDX 
It also happened to me to load svg and see them in the wrong position.

The cause is that the SVG format provides dozens of variability and often the programs used to generate the files apply their own custom standards. The code I use for reading SVG files (it is not my code) tries to handle all these probabilities, but always runs out of something.

If you provide a very simple sample file (ie. a square rectangle) that show the issue I can try debug the cause.

bbloss, 2018-11-20:
A tidbit that might help from my very short laser experience - I sometimes create my SVGs in Inkscape.

Inkscape has the concept of a page upon which you draw. You can draw outside the bounds of the page.

When importing my Inkscape-created SVGs into LaserGRBL the location of the drawn portion within my laser-bed are dependent upon their location to the page origin (lower left in my case).

closed 2018-12-06 by arkypita