arkypita/LaserGRBL #463: weak laser area preview

unkaMoMo, 2018-10-19:
Litefire has this great feature which turns on the weak laser beam and outlines the work area giving a chance to be certain where the burn will be on the workpiece.

StuartB4, 2018-10-20:
This Custom Button draws a box around the image to be engraved with low laser power.
Use this after you have prepared your image for burning.

[image]

[image]

This is the GCode, you can copy this and paste it in to the GCode box.

M4 
S5
F300
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

Explaination of code.
M4 Switches Laser ON.
S5 is the power setting. This can usually be anything from 0 min power  to 255 max power.
F300 is the speed setting and can be changed to suit you.
The next 5 lines draw a box around the image engraving area.
The next line moves the laser to the Home position at a speed to suit you.
M5 switches the laser OFF.

StuartB4, 2018-10-20:
Another useful button is a 2 State Button.

[image]

GCode for box 1:
G0 X[left+width/2]
Y[bottom+height/2]
G1 M3 S1 F900

GCode for box 2:
G0 X0 Y0
M5

Tooltip text:
Click to find center of image. Click again to switch laser off and return home.
M5

unkaMoMo, 2018-10-20:
That worked perfectly. Thanks for the quick reply. I think  I've found the software to use from now on.

closed 2018-10-20 by unkaMoMo