arkypita/LaserGRBL #464: Laser grbl zero position

dfox1787, 2018-10-20:
Can the zero position be set the the center? if not it would be really good.

Effer, 2018-10-23:
Also me prefer start laser engraving from center.
Just a propose, what about add an option in the "Target Image" form?

[image]

[image]

When checked the option keep adjusting the X and Y offset to the center (-Width/2 and -Height/2)

dfox1787, 2018-10-23:
Yeah that would be excellent. Lasergrbl image burning is way better than laserweb but laserweb offers some features that i still need for my projects.

Effer, 2018-10-24:
I have a fork [here](https://github.com/Effer/LaserGRBL/tree/masterEffer) with the small change necessary. I'm not sure about a pull request, probably it can be managed better by @arkypita .

StuartB4, 2018-10-24:
Hi effer.
Would'nt mind trying that.
I downloaded it but don't know how to install it, theres no installer.

Effer, 2018-10-25:
I'm sorry @StuartB4 and @dfox1787 but I'm not going to provide executable or installer, only the Visual Studio solution.
You then need to open the project with VS2017 Community Edition and compile from there. That's some work to do, just to test this small feature so maybe it doesn't worth at all.
I've mainly posted here to discuss the best approach to solve the issue.

StuartB4, 2018-10-25:
Hi Effer.
Thanks, thats fine but where to start. I tried 3 different compilers last night and got nowhere.
I've just tried to install  VS2017 Community Edition again this morning and during install it comes up with another load of stuff to install.
Any idea which ones to install ? I've never had to compile anything before so not really sure what to do.
[image]

arkypita, 2018-10-25:
Hi @Effer 
I am not ignoring your proposal but I have too few time now for the project.
Look a good solution but you can also obtain it with a custom button that set WCO to center of image

arkypita, 2018-10-25:
G92 X[width/2] Y[height/2]

You will not see the axis crossing your drawing, but the cursor go to center of image and coordinates takes 0,0 as current position

Effer, 2018-10-25:
Hi @arkypita the WCO way is probably the best approach to solve the issue, and it involve no cost on develop. So I agree for now just use this @dfox1787 .
I'm sure it's already be suggested to introduce some basic editing on the gcode from the main interface, for example allowing align the work (top,center,left,right) by altering the gcode.

@StuartB4 check the "Net Desktop Development", it should be enough.

dfox1787, 2018-10-25:
interested if anyone could come up with a button to change the WCS. Im happy with lasergrbl image dithering and if i could change wcs and also stop aspect ratio auto changing x when y is changed then id pretty much use lasergrbl all the time. 

But these are features are perhaps specific to me.

Effer, 2018-10-25:
Hi @StuartB4 I was referring to the fact I feel easier to center the engraving job in the working piece simply finding it's center, so aligning the machine it's just straightforward.
Starting burning from center have little sense in this type of filling pattern. Maybe using a spiral path (circular,elliptical or squared) would fit your needs.

StuartB4, 2018-10-25:
Ok, sorry I misunderstood.
I use this custom button to find center of the image after preparing it. Press it once to find the center, then center the workpiece under the weak laser dot, press it again and it returns home ready for burning. 
[image]

dfox1787, 2018-10-25:
> interested if anyone could come up with a button to change the WCS. Im happy with lasergrbl image dithering and if i could change wcs and also stop aspect ratio auto changing x when y is changed then id pretty much use lasergrbl all the time.
> 
> But these are features are perhaps specific to me.

can you send me the compiled version? Will this need doing every time there is a update?

closed 2018-11-19 by arkypita

maillemaker, 2019-01-12:
Hi.  I am a new LaserGRBL user, so forgive me if I am not understanding all of this.
I also would like a means to identify the center of the burn location, so that I can position the target material at that location.
Is this currently possible without a custom complile?

StuartB4, 2019-01-12:
HI Maillemaker.
Yes it should work.
You must click the Connect button first for it to work.
When you click it the first time it will move the laser to the center of any image 
you have loaded and prepared for burning. 
It will do it with low power laser so it does'nt burn but you can see where it is.
Click it a second time and it will return the laser to X0,Y0 position and switch off the laser.

You can copy this text and paste it
in to the first GCode box.

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

Copy and paste this text in to the GCode 2 box.

$32=1
G0 X0 Y0
M5

Add a Tool tip that will show up when you hovver over the button with the mouse.
Select Idle.
Click Save.

dfox1787, 2019-01-13:
Its start from Center i want to achieve. Most other programs allow you to choose top right bottom left center etc.