arkypita/LaserGRBL #465: LaserGRBL codename Rhyhorn

arkypita, 2018-10-21:
Hi all!

I have released a new version of LaserGRBL that include **centerline vectorization**.
This feature was requested by a lot of user and complete LaserGRBL with a "must have features" for all who need a simple way to engrave images.

This release was released as a pre-release so it is not automatically updated. 
To those who want to test it I recommend downloading the installer from: [here](https://github.com/arkypita/LaserGRBL/releases/download/v3.0.0/install.exe)

StuartB4, 2018-10-22:
> Yes, it is intentional. After talk with some user I decide it was better to
> remove a go-home instruction on abort that could be a problem.
>
> Think about at CNC... if you have the tool down inserted in the material,
> and you move it to 0,0 you will destroy the tool and the material. Better
> stop in place and let the user to decide what to do

Ok. Yes I see your point, good idea.
Thankyou.

aMirMOsavi, 2018-10-29:
Hi **MR**
The new **Centerline** feature does not work
It only shows the _progressBar_ but nothing happens
:(

arkypita, 2018-10-29:
Hi @aMirMOsavi 
Can you try to reinstall LaserGRBL? In a first moment I have released a first version with a bug (now solved).

If nothing change:
LaserGRBL use an external executable (autotrace.exe) located in C:\Program Files (x86)\LaserGRBL\Autotrace for centerline tracing. Please check if is not blocked by antivirus. At last, try to run LaserGRBL with administrator privileges.

let me know

aMirMOsavi, 2018-10-30:
**Thank you** for Your answer
The problem was **solved** - my installation _place_ was another drive.
I think it's **better** if it finds the **autocenter.exe** file from the application location
I mean, for example
`C:\Program Files (x86)\LaserGRBL\Autotrace\Autotrace.exe`
change to:
`.\Autotrace\Autotrace.exe`

closed 2018-12-06 by arkypita