arkypita/LaserGRBL #466: Linear? (overide settings)

Alx-I, 2018-10-21:
Hello, what is this parameter?

(sorry, can't find it anywhere...)

StuartB4, 2018-10-26:
Can you explain a bit more please.

Alx-I, 2018-10-30:
Hello,

Here are some more details->

Hello,

In the overide settings (during the job is working), we can adapt the power, the speed, and the linear parameter. 
For the power and speed it's logical, but I don't understand the linear.

I also don't understand what is the difference between the 3 first Grayscale options:
-simple average
-weight average
-optical correction

LaserGRBL is really great, but there is not a lot of discussions about it on forum, and the doc is not enough to handle it perfectly.
Thank you for your help!

arkypita, 2018-10-31:
> For the power and speed it's logical, but I don't understand the linear.

Linear: speed of G1, G2, G3
Rapid: speed of G0

-simple average
-weight average
-optical correction

> I also don't understand what is the difference between the 3 first Grayscale options:

Each pixel of a colorfull image has a RED-GREEN-BLUE triple of numbers.
Converting color image to grayscale require to convert this triple of number to a single number.

Simple average do the averege of them, weight average gives more importance to green and less to blue, and optical correct try to reproduce the human "grayscale" vision.
```

if (formula == Formula.SimpleAverage)
{
	RedFactor = 0.333F;
	GreenFactor = 0.333F;
	BlueFactor = 0.333F;
}
if (formula == Formula.WeightAverage)
{
	RedFactor = 0.333F;
	GreenFactor = 0.444F;
	BlueFactor = 0.222F;
}
else if (formula == Formula.OpticalCorrect)
{
	RedFactor = 0.299F;
	GreenFactor = 0.587F;
	BlueFactor = 0.114F;
}
```
The specific numbers in the question are from CCIR 601. see https://en.wikipedia.org/wiki/Grayscale for more information.

However... you don't need to know how and why they work to use LaserGRBL. Just try one or another and see what will look better with your specific image.

closed 2018-10-31 by arkypita

Alx-I, 2018-10-31:
Ok nice! Even if I don't need to know to work with, I like to know what I do ;)
I have only tried with already grayscale pictures, that's why I didn't see difference between the 3 types.
I'm going to test it deeper.

Thank you!