arkypita/LaserGRBL #467: Windows 10 .net framework 3.5 won't install to launch GRBL

dylan1987, 2018-10-22:
Hi, Just downloaded this program for my engraver. I have windows 10. When I try to start the program, I get a prompt to add a program (.net framework 3.5). I choose "Download and Install this feature" but then get an error message: The system cannot find the file specified
Error code 0x80070002

StuartB4, 2018-10-22:
Hi Dylan.
Try this first.
Watch this video from 4 minutes 24 seconds just to check 3 of Windows Services are enabled correctly.
There are other ways to fix it but if this one works it's the easiest one to start with.
https://www.youtube.com/watch?v=ba6xFpCJIRg

dfox1787, 2018-10-23:
easy fix. download the W10 iso from ms and point the .net to sources folder.

fantilator, 2018-10-24:
I needed I have a .net 3.5 offline installer for Windows 10
You can download it here.

[.net Framework Offline Win 10](https://filetank.synology.me:5001/d/f/450863847438295041)

closed 2018-11-19 by arkypita