arkypita/LaserGRBL #470: LaserGRBL tags

fantilator, 2018-10-24:
What are all tags defined in LaserGRBL?
[UP]
[Down]
[Left]
[Right]

Are there some more?

Is there some solution to have the current X or Y coordinate in a custom button?
I'd like to make a button to find the center of the workpiece.
dual button.
Gcode1:
G92 X0   : Zero X axis
Gcode2:
G0 X/2   : devide current X coordinate by 2 and move to there
G92 X0  : Zero X axis again. this is my middle point of the workpiece in X axis.

Do the same with Y axis.

StuartB4, 2018-10-24:
This custom button finds the center of workpiece.

GCode for box 1:
G0 X0 Y0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M3 S1 F900

GCode for box 2:
M5
G0 X0 Y0

Tooltip text:
Click to find center of image. Laser ON. Click again to switch laser OFF and return home.

arkypita, 2018-11-19:
Implemented, will be available in the next LaserGRBL version.

As because you can set a coordinate offset, LaserGRBL will provide variables for Machine Position, Working Position and Coordinate Offset.

List of available variables: http://lasergrbl.com/usage/custom-buttons/

closed 2018-11-19 by arkypita

garthanstey, 2018-11-27:
On the subject of Custom Buttons, I was wondering if anybody had asked for customised images for the buttons. and maybe you should have to second click a double push button before another button can be pressed or the option. Just a thought.