arkypita/LaserGRBL #474: framing

Oldshed, 2018-10-26:
I have entered the g-code to frame an image.  Where does the [left]  [right]  numbers come from?

Oldshed, 2018-10-26:
> From the image size.
> You should have an image loaded in first.
>
> This if from my custom button.
> I load the image and prepare it for burning, then press the Framing custom
> button and it moves the x,y stepper motors around the outside of the image
> whatever size it is.
> This does it with the laser on low power (S10) at a speed of 300 mm (F300)
> per minute.
>
> M4
> S10
> F300
> G0 X[left] Y[bottom]
> G1 X[right]
> G1 Y[top]
> G1 X[left]

sorry for being so dense, but where do the values for  left,  bottom,
right,  top,   come from?  When I enter those g-codes lasergrbl puts a
little slash circle at the left and nothing happens.  My image is loaded
and system at idle.  What am I missing?

StuartB4, 2018-10-26:
Where are you entering them?

arkypita, 2018-11-19:
left, bottom etc are available only if entered in custom buttons.
http://lasergrbl.com/usage/custom-buttons/

closed 2018-11-19 by arkypita