arkypita/LaserGRBL #475: Improving the scarring of the laser lens

wilu9500, 2018-10-27:
Hello, I am expanding my machine. I have increased the machine's working area.

[image]

I would like to know your opinion because I have a 5.5 Watt blue laser with a wavelength of 405 nm. In the set there was such a lens that focused the laser. For M9 thread.

[image]

It seems to me that there are better lenses for lasers of this type. I would like to know if you have experience with improving the focus of a laser beam of this type.

StuartB4, 2018-10-27:
You could always join the forum here:
[Laser Forum.](https://laserpointerforums.com)
They have loads of help and experience .

Focus3, 2018-11-02:
Good day,

G2 lenses seem to be top choice. Just search for a G2 lens with the right wavelength (405 nm should be in the range).

Focus3.

closed 2018-11-19 by arkypita