arkypita/LaserGRBL #477: Some lines simply vanish while being exactly the same as other that didn't.

RFVasco, 2018-10-30:
I've been using Autocad to draw images to engrave with LaserGRBL, but when I load those images on the program, sometimes some lines simply vanish without any criterion, considering I only drew one line and the others where copied. Any ideas how to solve this? I've already thickened the lines on autocad, and while it solved a few of the issues, some others appeared.

[image]
 This image works just fine, but when I try to mend 2 images or more in one, some lines that were fine vanish.

[image]
The line below "Echeveria" doesn't appear on the program using this image.

[image]
None of the midle lines appear with this image.

StuartB4, 2018-10-30:
Have you tried adjusting the Brightness, Contrast and White Clip.
I tried with the image you posted and it seems ok with the settings I used.

[image]

[image]

RFVasco, 2018-10-30:
Believe it or not, I tried everything, but I don't know the numbers exactly. What was bugging me was that sometimes, with the same configs, some lines would just disappear. Appearently those numbers of yours solved and now all images are working just fine. Thanks for the help (:

arkypita, 2018-10-31:
@RFVasco as you know, LaserGRBL does not directly support dxf file (for now) so, I think you export your draw in raster image format (.png or .jpg) and then import them in LaserGRBL.

What append in LaserGRBL is that regardless of the exported image resolution, the image is streched (downscaled or upscaled) to fit the required "line/mm * size" resolution.

What I mean: if your export is 4606x1654 (as your last picture posted) and you want to engrave it to a 
23x8cm @ 7lines/mm your image is downscaled to 1610x560 px (230*7 x 80*7) because this is the pixel matrix that is plotted by the laser to respect your request of 23x8cm @ 7lines/mm. 

This downscaling procedure will naturally lost some details.

What I see is that your third picture has very thin lines. about 2pixels of width, and both are "not so black" as they should be, so when scaled the "anti aliasing" algorithm will smooth/fade them and if downscaled too much they will be lost.

[image]

arkypita, 2018-10-31:
This effect could be compensated by increasing contrast, but I suggest a better way that could help too.

When you export your image you can manually compute the optimal export resolution. According to previous example if you need engraving 23x8cm @ 7lines/mm export the image at 1610x560 so LaserGRBL doesn't need to resize it.

closed 2018-10-31 by arkypita