arkypita/LaserGRBL #481: M4 command don't work.

EpoHub, 2018-11-03:
Hi
M4 command don't work. Program always generate code with M3, I open saved gcode and here also M3 instead M4.

StuartB4, 2018-11-04:
Try this:

[image]

After clicking the Write button you can exit Grbl Config.

If you don't have $30, 31 and 32 in the list then you have an old version of GRBL and need to upgrade to version 1.1.

StuartB4, 2018-11-04:
Thats really strange. 
Does it only do it with SVG files.
I ask because yesterday I posted this to Arkypita in the SVG thread.
https://github.com/arkypita/LaserGRBL/issues/451
Last entry in the thread.
But it's opposite to your problem.
When I change it to M3 it still writes the M4 command in the code.

EpoHub, 2018-11-04:
> Thats really strange.
> Does it only do it with SVG files.
> I ask because yesterday I posted this to Arkypita in the SVG thread.
> #451
> Last entry in the thread.
> But it's opposite to your problem.
> When I change it to M3 it still writes the M4 command in the code.

Yep you right I vectorize png and program generate code with M4.

arkypita, 2018-11-04:
Maybe you are both right. SVG to gcode algorithm is something I have copied so not totally under my control/knowledge, I have included it in LaserGRBL but it is not totally well integrated.

Moreover in v3.0.0 i had to restore a previous version of this code because the needs to integrate it with centerline vectorization.

I will verify it asap
Keep in touch from the next version.

StuartB4, 2018-11-05:
Ok Arkypita.
Thanks very much for the explaination.

closed 2018-11-06 by arkypita

arkypita, 2018-11-06:
fixed

StuartB4, 2018-11-06:
Wow, that was quick.

StuartB4, 2018-11-07:
Thanks very much. 
Tested it a few times and it works absolutely fine now.

commit cb41da4f (2018-11-06): fix some issue with SVG import

fix #481