arkypita/LaserGRBL #486: Cursor position worng 3.0.1

RDS356, 2018-11-09:
In this version , the cursor follow from machine position.
In the lastset version the cursor followed the WORK POSITION.
Why do you change it ?
[image]

arkypita, 2018-11-19:
Actual LaserGRBL version has this behaviour:

- First Line: always show Machine Position X, Y and Z (Z is optional, shown only if relevant)
- Second Line: optionally show Work Position X, Y, and Z (if relevant). The whole line is optional, shown only if there is a working offset active.
- Third Line: optionally show feed speed and power (F and S) if reported by grbl status report.

I don't remember how it was in older version but your screenshot of 3.0.1 is coherent with designed behaviour.

arkypita, 2018-11-19:
> Why did you change it ?

Because people that does not use working offset now does not see an useless line

closed 2018-11-19 by arkypita