arkypita/LaserGRBL #490: Root element Is Missing

Megistus1, 2018-11-15:
Thank you for such a Great program.
Root element Is Missing Box keeps popping up whenever I try and use the Centerline setting.
Same on two different computers running Windows 10.
Thanks
[image]

BallscrewBob, 2018-11-16:
Also seeing that.
Does seem to work very well with less segments to calculate.
Actual slider settings dont seem to make any difference with the error though.

StuartB4, 2018-11-16:
@BallscrewBob .
Are you using Windows 10.
I've tried it loads of times with different settings on Windows 7 and it's not produced that error.

arkypita, 2018-11-19:
Centerline use an external tool called autotrace located in C:\Program Files (x86)\LaserGRBL\Autotrace\autotrace.exe to transform the raster image to a SVG vector file with centerline paths.

After that LaserGRBL load the SVG produced by autotrace and convert it to GCode.

The error shown highlights that the first step (jpg to svg by autotrace) has produced an empty file (without svg/xml content on it).

In my environment i never seen this issue. Please provide a step-by-step description on how to reproduce the error, type of your pc and available ram, and provide the file you are using for test.

arkypita, 2018-11-19:
also be sure that autotrace is located in this folder and  is not blocked by antivirus etc

vale67, 2018-11-19:
I tried to run autotrace but I get this error message (O.S. win 10)
[image]

arkypita, 2018-11-19:
is your OS 32 or 64 bit?

arkypita, 2018-11-19:
> is your OS 32 or 64 bit?

closed 2018-11-19 by arkypita

BallscrewBob, 2018-11-19:
Windows 7 x64.
Will add location to exclusions in AV but it does work with less work to do so suspect a size/segment issue and not an AV issue.

myfordboy, 2018-11-19:
I have the same issue and cannot get centerline to work. 
I checked my C drive > program files and Autotrace is there. Clicking on it brings up the message
" Not comptable with the version of windows you are running, contact software supplier"
I am running windows 7 32 bit. Is there a solution?

arkypita, 2018-11-19:
> The problem could be here?

No. In this folder you can find saved settings, custom buttons, statistics and sessionlog (where lasergrbl records some diagnostic info).

I cannot test with 32bit but i guess the problem is on it.
As soon as i have time i will do some test on a VMs

arkypita, 2018-11-19:
autotrace is a command line program. Its behaviour, if launched from double click on it is to open and close immediately, but no error should be shown.

myfordboy, 2018-11-19:
Further to my previous comment,  not able to use centerline, I have tried running on another Win7 laptop that is 64 bit and it runs OK.
The 32 bit laptop is my workshop one and I would really like to use that one if it is possible.

StuartB4, 2018-11-19:
Just a bit of info.

Just tried an intricate line drawing in centerline mode and it was fine.
I then adjusted the maximum travel distances for X and Y in GRBL configuration settings
and tried it again.
Usually my machine has a maximum travel of 40 x 40 mm.
I changed it to maximum 1000 mm's in GRBL.
I used the same image and set the W and H settings in the Target Image settings to 950 mm's,  and eventually after about 15 minutes of the bar going backward and forwards along the bottom of the window, I got the same message.

After that I worked backwards lowering the W and H bit by bit settings from 950 and it reproduced the error every time until I got as low as W=77 and H=77. Then it worked fine with no error. 
I have Windows 7 Pro 64Bit, 4 Gig Ram.

[image]

arkypita, 2018-11-19:
I think we are seeing the effect of multiple causes.

**Out of memory or root element is missing message**
Seem to be caused by image size. One problem is that autotrace prefere .bmp file format, a format that needs huge amount of memory to be managed. Another problem is again on file size that seem autotrace have some limit on it.

In this cases you see centerline working in preview and an error while producing final ouptut.

**unresponsive or totally not working**
as signalled by @vale67 or @myfordboy, look like to be related on 64/32 bit version of the OS

arkypita, 2018-11-19:
i am working on it

arkypita, 2018-11-22:
done a lot of work on centerline in v3.0.3
now can work up to 600x600mm size

not solved issue with 32bit

garthanstey, 2018-11-27:
Include me in the 32bit issue. My wife won't let me take her laptop into my den.

vixory, 2019-02-16:
доброго времени  суток, у меня проблема с функцией "centerline" выдаёт ошибку.  
операционная система стоит Windows 10 на ноутбуке. 
версии LaserGRBL почти все перепробовал, результата не дало.
[image]

myfordboy, 2019-02-16:
I have contacted the developer on some issues I had with the centreline
feature. I initially ran it on a laptop with a 32 bit OS and could not get
the centreline to work. I also have another laptop running 64 bit and it
runs well on this. Both are win7. At the time the issue with 32 bit was not
resolved.
Look here https://github.com/arkypita/LaserGRBL/issues/490
You can ask questions here
https://github.com/arkypita/LaserGRBL/issues

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>