arkypita/LaserGRBL #496: Rhinoceros, Dxf, taglio 1:1

filiqqo, 2018-11-18:
Ciao, complimenti innanzitutto complimenti per il software. L’unica pecca è che non è possibile importare nessun file dal cad Rhinoceros. Secondo me la limitazione che fa scegliere software a pagamento invece che LaserGRBL è solo il fatto di non potere importare file DXF che sono esportabili da qualsiasi tipo di cad.... Esiste una soluzione per disegnare in Rhino e tagliare con LaserGRBL ? Grazie

StuartB4, 2018-11-18:
È possibile convertire i file DXF in SVG e caricarli in LaserGRBL.
https://cloudconvert.com/dxf-to-svg
Basta fare clic su "Seleziona file" (Select File)
Quindi fare clic su "Seleziona dal mio computer" (Select from my computer) per caricare il file DXF.

Tradotto dall'inglese con Google Traduttore.

filiqqo, 2018-11-18:
Funziona, ma una volta importato in LaserGRBL il file .SVG non e' modificabile la dimensione e la posizione  e non è possibile creare un riempimento fillet

StuartB4, 2018-11-18:
Sì, l'ho notato quando l'ho provato io stesso.
Inoltre alcune delle immagini che ho provato erano troppo grandi e scomparivano dallo schermo. Mi dispiace per quello
Ci sono altri convertitori in linea, ma ne ho provati altri 2 con risultati simili.

arkypita, 2018-11-19:
Ciao. L'import dei file DXF è già in dev roadmap #64 e verrà implementato compatibilmente con altre priorità della mia vita (in poche parole quando avrò tempo e voglia di farlo, senza possibilità di predizione).

Nel frattempo non so suggerirti un tool di conversione da DXF a GCode (eviterei il passaggio per svg) ma sono sicuro che se cerchi in google trovi qualcosa che fa al caso tuo.

closed 2018-11-19 by arkypita