arkypita/LaserGRBL #497: Laser does not turn off, problem solved

David7820, 2018-11-19:
Hello everyone, first I apologize because this is a google translation.
I decided to write this after solving my problem, even though I do not have enough knowledge about grbl and I am new to the cnc world ..., maybe someone will save the time I lost installing my laser (a lot) and run LaserGRBL. The problem I had is that the laser simply did not turn off at any time, in any way, with all the software installed correctly. If someone has this problem with a laser like mine (built-in pwm), I finally solved it by simply connecting a ground wire (0V from the laser power supply) directly to an Arduino Gnd pin. Before I had it connected to the general Gnd of my homemade machine (in this way, I do not understand it, but it does not work even though they were all connected, including Arduino).
Thank you for being able to use this software.
I hope to help.
[image]

OldShark, 2018-11-21:
Hi David. U suggests something like this?

[image]

David7820, 2018-11-21:
Hi, yes to the arduino directly, to the common gnd of the cnc it did not work for me. Although I used the power supply that comes with the laserThere is not much information about these lasers, I almost give it as impossible.

OldShark, 2018-11-21:
I think it's not impossible. I already detected the problem in mine. I'll put the image at night. I connected the ocilloscope to pin 11 (Z +). the pulses strike right with the GRBLLaser tracing the image. The voltage sent is okay for laser firing in dark areas. but this without connecting pin 9 in the laser driver. When connecting to the laser driver, the voltage on pin 11 drops by half. I'll do a test. I will unsolder the SW-LD62 cooler (LD driver). I think he's actually draining something. This being real, I'll use a source just for it. otherwise I will put a set-up on pin 11 to raise the voltage for sending to the TTL.

OldShark, 2018-11-21:
Do you have a multimeter to be able to measure the voltage at pin 11 (Z +) connected to the LD and without connecting?

OldShark, 2018-11-21:
And sorry my poor english. ahahahahaahaha

David7820, 2018-11-21:
Sorry, not understand the question ¿LD?
, My english is worse

OldShark, 2018-11-21:
Are you french?

David7820, 2018-11-21:
From Spain, Canary Islands

OldShark, 2018-11-21:
Ok. Let´s go. Google translate:
¿Tiene un multímetro para poder medir el voltaje en el pin 11 (Z +) conectado al LD y sin conectar?

OldShark, 2018-11-21:
soy brasileño.

OldShark, 2018-11-21:
pero una duda que no entendí. ¿has conseguido solucionar el problema con ese láser y el cnc shield?

David7820, 2018-11-21:
Si resolví el problema, no tengo cnc shield, drivers externos para los motores conectados directamente a arduino.El problema es que la fuente de alimentación de estos laser no tienen una toma de tierra.Esto es lo que midoConectado  0.046vTrabajando  hasta 3.2vSin conectar 0.01vNo se si es lo que me pidesPerdón se me fue en español

OldShark, 2018-11-21:
creo que es exactamente el problema. la noche llegando del trabajo voy a poner las imágenes de las mediciones que realicé

OldShark, 2018-11-21:
algo está drenando la corriente en el perno 11

David7820, 2018-11-21:
Please to meet you

OldShark, 2018-11-21:
sorry . corrente = cadena

OldShark, 2018-11-21:
and nice to meet U2

OldShark, 2018-11-21:
Go home now. when i came home put some photos

David7820, 2018-11-22:
> buenos días David. desconectar el refrigerador no ha cambiado en nada. la caída de la señal permanece. voy a intentar poner un regulador step-up. Vamos a ver. gran abrazo

Saludos

closed 2018-12-06 by arkypita