arkypita/LaserGRBL #502: 10K User! We are a community

arkypita, 2018-11-22:
Hi all! I am happy to announce that LaserGRBL has had over 10000 unique users from all over the world in the last month. This is a big achievement.

To celebrate I decided to open a [facebook page of the community](https://www.facebook.com/LaserGRBL) where I invite you to exchange photos of your work done with LaserGRBL, lasering tips and tricks, schematic and photos of your DIY engraver etc.

I will not do software support on this page, support channel will remain ghitub.

arkypita, 2018-11-22:
My first thanks goes to @StuartB4 that actively contribute to software support.

Another special thanks to all those who have made a small or large donation. I appreciated. Obviously this thanks goes out to all of you who use LaserGRBL and have made the community so numerous.

StuartB4, 2018-11-23:
Wow!! 10,000 amazing.
Congratulations Arkypita.
It just shows what a success the program is.
The support you provide and the time and effort you put in to it is excellent considering it's free.
The program gets better as time go on and is a pleasure to use.
Thanks for the mention, I'm always happy to help if I can.
I don't always get it right, but we all make mistakes.

Good luck to you and for the future of the program.

Mikalay31, 2018-11-23:
I will help your program to have even more users) I advertise you among the Russian-speaking population! I wish you even more users)