arkypita/LaserGRBL #503: Stop button must to be M30 , not m5.

RDS356, 2018-11-23:
Hi..
In my thought , when the stop button is pressed the command sent must to be M30(end program and reset all outputs like m3,m4,m8).
I told about this before, and I'm sorry for anything.
Congratulations for 10k users.

arkypita, 2018-11-26:
I am sure that actual behaviour is the better one. Actual LaserGRBL version stop sending code and push a single M5 (laser off). Grbl run the last 3-4 gcode command enqueued and then perform laser off and stop moving.

The way to reset is in the hand of the user. If M30 si good for you, could be not good for anyone.

> In my opinion the abort button should not exist, because its behaviour is not so clear.
> I prefer to use pause button (red hand, bottom left of UI) and grbl reset, but a lot of people ask me for a abort button.

However I will experiment a few with M30 to see its behaviour

arkypita, 2018-11-28:
> Well, if you want stop the machine all outputs must turned off. M30 make this, m5 just turn off the m3 output.

Ok, now it is more clear, but I never found a consistent documentation of M30
Moreover seem to have some issue on older versions: https://github.com/grbl/grbl/issues/681
At the very least I have to differentiate the behaviour.