arkypita/LaserGRBL #504: One more time about # 123, # 245b

LucAn212, 2018-11-23:
Hi,
This is excellent program. 

Two proposals:
(I have modified the program and used these options for several months)

A. Dimensioning of the image based on the resolution [pxl]. # 123, # 245b 
This option is basic to me.

On the "Target image" screen, I additionally inserted:

1. "Autosize ON/OFF" switch. If it is OFF, the program works like the original one. 
In the ON option, calculates the image size based on the resolution and enter it into the current "Size" W, H fields (with a change lock or manual entry)
W[mm] = 25.4 * W[pxl] / DPI

2. I added the DPI field to manually enter the resolution. The default (starting) DPI value = 300  (DPI>0)
If you enter zero, the field will return to the default value.
DPI can be entered regardless of the "AutoSize" state. Confirm Enter.

B. Save Program -> name.nc = name.jpg    # ???  (there was a request)
(Additional code in GrblCore.cs)

--------------------------------------------------
Both proposals are already included in the files, which should be replaced in the original program v.3.0.3 ! In older versions it will not work, in the meantime @arkypita updated these files.

I'm not a programmer !!! 
You can probably do it more professionally and elegantly. For me the most important thing is that it works. Check, make changes and use.

I wish you perseverance in further improving the program.

------------------------------------
* I'm sorry.Text translated by Google Translator.
[A+B.zip](https://github.com/arkypita/LaserGRBL/files/2611350/A%2BB.zip)

arkypita, 2018-11-26:
I'll take a look to your code.
It's one of the things I have in the roadmap

arkypita, 2018-11-26:
> B. Save Program -> name.nc = name.jpg # ??? (there was a request)

good hint

LucAn212, 2018-11-26:
For those who want to see how it works.
[LaserGRBL 3.0.3 _xx.zip](https://github.com/arkypita/LaserGRBL/files/2615961/LaserGRBL.3.0.3._xx.zip)

arkypita, 2018-11-27:
how does DPI/Autosize interact with line/mm engraving resolution? it is only used for compute image size or it is used as engraving resolution?

LucAn212, 2018-11-27:
Only for the dimensions of the final image.  Exactly the same if you manually enter Size W or H. in the original program.
With the resolution of engraving has nothing to do. It does not affect other program functions.

More precisely:
It's just a mathematical calculation in   (ConvertSizeAndOptionForm.cs)

 private void CheckBox1_CheckedChanged(object sender, EventArgs e)
 private void PxlDPI_CurrentValueChanged(object sender, int OldValue, int NewValue, bool ByUser)

Two functions because it can be
- DPI change with Autosize OFF
- DPI change with Autusize ON

The mTrueOriginal.Width value is taken from   (ImageProcessor.cs)

        public float getImageWidth()
        {return mTrueOriginal.Width; }
--------------------
name.nc = name.jpg only the function has been changed (GrblCore.cs)
public void SaveProgram()

arkypita, 2018-11-27:
done in development (master) branch
try download source and let me know

i prefer not use 300dpi as constant value but detect resolution from file information

LucAn212, 2018-11-28:
Perfect!
Maybe three notes.
1. At my place default DPI = 300 - intentionally. Not only resulting from my needs, but also for safety.

- not all files have DPI in their properties. You wrote about it yourself in # 123. Then what? I did not have such files to check.

- on an example: I have model plans (on paper) in 1: 100 scale on A2 sheet. This is too big a picture.
Scans (DPI = 300) -> file.jpg - divide (cut) into smaller -> detail_1.jpg, detail_2.jpg ... detail_80.jpg
They all have DPI = 300, so they retain the 1: 100 scale and all elements match.
I clean each of these files from impurities in the graphic program. Some programs change DPI without my knowledge to 96. Engraving at such DPI will give an element over 3x bigger. If I have permanently set DPI = 300 then the DPI of the file does not matter. At the end I will always get the right size. (I checked)

This is not a problem, I can customize and compile it myself

2. There is no fuse for DPI = 0

3. Entering DPI without Enter (like Size W). I do not know how to do it.

Good night.

arkypita, 2018-12-01:
added in v3.0.4 let me know

LucAn212, 2018-12-02:
Thanks a lot

closed 2018-12-02 by arkypita