arkypita/LaserGRBL #507: ngc file format

ALDEEN19, 2018-11-25:


ALDEEN19, 2018-11-25:
I think adding *.ngc  file format would make it more versatile.
I also couldn't find buttons for moving Z axis machine manual control.
Thanks a lot.

StuartB4, 2018-11-25:
Go to
https://github.com/arkypita/LaserGRBL/issues/306

Scroll down a fair way and look for a post by 1estrider.
He has a custom button for manual movement, Z up and down.

If you are not sure how to make a custom button let me know here and I will explain.

ALDEEN19, 2018-11-25:
Sure I already did created custom buttons for the Z axis movements
right just after I wrote my comment, so thanks for your prompt reply.
I am looking forward to have the ngc file format included among the
other available gcode formats.
Thank you again, I really appreciate your pleasant effort.

StuartB4, 2018-11-26:
Ok no problem, glad you got it sorted.

arkypita, 2018-11-26:
.ngc is only an extension for a text file containing gcode?

ALDEEN19, 2018-11-26:
ngc file extension is normally generated by "gcode tools" embedded in
Inkscape and it is usually a gcode file just like .nc extension.
In fact I have tried using this particular ngc extension in your nice
LaserGRBL application by choosing any svg file and then I change its
svg extension manually by typing ngc instead of svg and then Enter,
and it works always perfect just like any other gcode file.
Cheers!

arkypita, 2018-12-01:
added in v3.0.4 let me know

closed 2018-12-01 by arkypita

ALDEEN19, 2018-12-02:
Yes ! .. ngc file format now is viewable and nicely working.
Thank you very much.