arkypita/LaserGRBL #509: Task hanging.

pleuris666, 2018-11-26:
every once and aawhile my Laser keeps hanging on a line. It engraves normally until a certain point and keeps lasering that single line.

is there something I can do about it?

StuartB4, 2018-11-26:
Have you made sure your x and y axis's rails are clean, and you have no cables too tight that would restrict the movement.

pleuris666, 2018-11-27:
Good morning. My engraver uses Vigotec engraver 2.01. And needs an encryption lock-USB key.(USB stick) I think the problem is in that when I use the program the program thinks the USB key is no longer inserted. I get the fault message: wrong, no encryption lock. I get the same message when I start up the program when the US key is not inserted in my computer. The software will not start up and gives the same message. Do you have an idea how I can work arround it? What I tried to do so far:Removed all unessential USB devicesTried different USB ports for both the engraver and the USB encryption lock USB stick. Firewall and VirusScan is OFFThe USB stick is recognized by windows 10 as USB key. I cannot acces it. Thanks Joris Gadellaa Verzonden vanaf mijn Huawei mobiele telefoon-------- Oorspronkelijk bericht --------Onderwerp: Re: [arkypita/LaserGRBL] Task hanging. (#509)Van: StuartB4 Aan: arkypita/LaserGRBL Cc: pleuris666 ,Author Have you made sure your x and y axis's rails are clean, and you have no cables too tight that would restrict the movement.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arkypita/LaserGRBL/issues/509#issuecomment-441788872",
"url": "https://github.com/arkypita/LaserGRBL/issues/509#issuecomment-441788872",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [arkypita/LaserGRBL] Task hanging. (#509)",
"sections": [
{
"text": "",
"activityTitle": "**StuartB4**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@StuartB4",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"arkypita/LaserGRBL\",\n\"issueId\": 509,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"arkypita/LaserGRBL\",\n\"issueId\": 509\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/arkypita/LaserGRBL/issues/509#issuecomment-441788872"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 416188032\n}"
}
],
"themeColor": "26292E"
}
]

StuartB4, 2018-11-27:
Is Vigotec engraver 2.01 the name of the engraver machine or the software ?
If it's the name of the software then you should ask the software developer.

closed 2018-12-17 by arkypita