arkypita/LaserGRBL #510: Problem of mismatch between fill and stroke lines

Phenixoid, 2018-11-27:
Hello!
I want to immediately apologize for my bad English - Google Translate.
My machine consists of a bunch of Win7x64, LaserGRBL v3.0.3, GRBL v1.1f, mechanics based on Anet A8. When burning, I encountered the problem of mismatch between fill and stroke lines.

I open the PNG file. I set the following parameters:
```
Resize			Smooth (HQ Bicubic)
Brightness		100
Contrast		100
White Clip		11
B&W			On, 50
Conversion Tool		Vectorise!
Vectorise! Option
Smoothing		On, 10.0
Filling			Horizontal
Filling Quality		12 lines/mm
Speed			65 mm/min
Laser Options		M4/M5 S 0-255
Size			150x27 mm
```
The image is part of the phrase. Everything looks good in the preview window. Burnout lasted more than 4 hours. At the output, I got the offset of the stroke lines relative to the fill. Moreover, on the "correct" positions is filling. Stroke lines have a negative offset; the closer to the zero coordinates, the more offset, by the end of the phrase being burned off, the offset is reduced.
What could it be?

arkypita, 2018-11-27:
can you upload a picture?

Phenixoid, 2018-11-27:
Of course. Here are the files. 
[image]

[123.nc.txt](https://github.com/arkypita/LaserGRBL/files/2618980/123.nc.txt)

Phenixoid, 2018-11-27:
I can not attach the photo of the result obtained at the moment. I think by the evening I can provide.

Phenixoid, 2018-11-27:
Photo
[image]
[image]
[image]

Phenixoid, 2018-11-29:
Greetings. I carefully watched the machine and tried to reproduce the problem. It turned out that the flexible cable channel along the X axis, with a considerable distance from the origin of coordinates, began to spring and push the carriage of the X axis closer to zero. I also found out that when burning filling lines, the engines are always in a state of motion. At the beginning of the burn-out of the stroke line, the head rises to the desired position and at this moment the engine stops and “releases”. The farther the carriage along the X axis, the more it will spring the cable channel back at the moment of stopping.
I conclude that I have a badly tuned GRBL. Your program has nothing to do with it. Sorry I thought for your software. The topic can be closed.

closed 2018-11-29 by Phenixoid