arkypita/LaserGRBL #512: Can't import images on older versions

jranavaya, 2018-11-28:
HI,

So I have been using LaserGRBL for a while now and been using the older 2.8.52 version fine. I recently upgraded to the newest 3.0.3 version and was experiencing some issues in my engravings. I wan't sure if it was the software or hardware so I decided to uninstall and try the 2.8.52 version again. This time it seemed ok until I got to the import image raster screen. I can set it all up fine but when I hit continue to set speed and size of engraving I get this error. 

This happens no matter which version I use, apart from the 3.0.3 version. Any idea what it is?

[image]

jranavaya, 2018-11-28:
Actually I have just checked again and seems every version works from the latest down to 2.9.2. Anything older than that it doesn't work and gives that error.

jranavaya, 2018-11-28:
> try to delete this file: C:\Users\YOUR USER NAME\AppData\Roaming\LaserGRBL\LaserGRBL.Settings.bin
> this reset the lasergrbl config and older options

Thanks, I tried that but it didn't change anything. Same error when I try version 2.8.50. 

> or better: can you zip and send it to me so I can reproduce the error and make my code stronger?
> my email: [info@lasergrbl.com](mailto:info@lasergrbl.com)

I emailed you the file :)

arkypita, 2018-11-28:
> So I have been using LaserGRBL for a while now and been using the older 2.8.52 version fine. I recently upgraded to the newest 3.0.3 version and was experiencing some issues in my engravings. I [...]

When you have updated to v3.0.3 the value of Width, Height, Offset X and Y, was updated to be floating point (with decimal point) instead of integral number. Then the values you have used was stored in config file (as last used values) as floating point. 

When you run an older version, the older version doesn't know nothing about this change, and the old code try to read them as integral number, so there is an exception. When you downgrade you should delete setting file (not always, but in this case you should).

jranavaya, 2018-11-28:
I already tried deleting the settings file you mentioned and I still got the error. This issue should not be closed.

arkypita, 2018-11-28:
I can reopen the issue but I can not help you much more because I can not reproduce the issue, even using your file. Just see if anyone else report the same

closed 2019-01-13 by arkypita