arkypita/LaserGRBL #513: No dimensions settings for SVG files?

thmsnhl, 2018-11-28:
Wenn throwing in svg files, there is no settings window for adjustments on the dimensions and fill method which forces me to stick to pixel files and use "vectorize" option.

Is this the expected behavior? 

LaserGRBL V3.0.3

arkypita, 2018-11-28:
> Is this the expected behavior?

Yes. 

> there is no settings window for adjustments

SVG contains size in its internal format, LaserGRBL read those info and use them. If you need a different size you can resize the document in your favourite SVG editor (I use inkscape).

> and fill method 

If you want filling you should create filling path in your svg and apply to your shapes/path. SVG is not a raster format, so LaserGRBL manage svg file as it contains instruction for engraving directly and respect them, so no filling option (for now).

closed 2018-11-28 by arkypita

LordPueblo, 2019-03-19:
Dear @arkypita you say 

> you should create filling path in your svg

Please let me know how to do it in Inkscape?

gisep, 2019-03-20:
Come on guys, you cand find it in www.google.it with a little effort.

https://wiki.evilmadscientist.com/Creating_filled_regions