arkypita/LaserGRBL #514: 32bit system problem

kefeur, 2018-11-28:
[image]
Hello!
The problem is a newer version of the 64bit system for generating centerline.
At my request it would work 32bit.
Thanks.

kefeur, 2018-11-28:
Error message:% 1 is incompatible with 32-bit or 64-bit versions of Windows. My machine has win7 32 bits.

arkypita, 2018-11-28:
This is a know issue. Unlucky the external program (autotrace.exe) i use to compute centerline is only available as precompiled exe for 64bit OS. I am looking for a solution but for now you should accept this limit

kefeur, 2018-11-28:
OK . Thank you forissues.

kefeur, 2018-11-28:
[image]
Can it work?

kefeur, 2018-11-28:
Martin Weber
martweb@gmx.net

kefeur, 2018-11-28:
http://www.inkscapeforum.com/viewtopic.php?t=11898

arkypita, 2018-11-29:
@kefeur I have tested the zip. Seem to work in a XP 32bit virtual machine.

The worst thing is that the executable is not autonomous, but it depends on all the libraries that are distributed with the zip. I would not like to fill my installation folder with all this files. I am looking if I can compile a stand alone autonomous version.

Meanwhile can you try to decompress the zip in your LaserGRBL installation folder, under autotrace subfolder, and replace actual autotrace.exe with the new exe and dll from zip? Let me know.

kefeur, 2018-11-29:
It does not work well. He writes "the value can not be null (zero).
[image]

kefeur, 2018-11-29:
Other autotrace for windows : 
[image]
What is it ?

kefeur, 2018-11-29:
https://github.com/PancakeBot/PancakePainter
https://github.com/autotrace/autotrace

closed 2018-12-07 by kefeur