arkypita/LaserGRBL #518: New Feature: Possibility of measurement scale.

StuartB4, 2018-12-03:
Would it be possible to add a measurement scale along the bottom edge 
and down the left side edge of the image window.

At the minute it shows the size of the image, but nothing in between
The image below was 2 seperate elements.
The text loaded first and then the picture added using Append File.

Selecting the correct placement settings in Target Image Offset settings for the second image
would be easier if you could see a scale along the bottom and down the left side.
A line with Incremental marks of 1, 2 or 5 mm's would be enough.

[image]

arkypita, 2018-12-03:
Like it!

arkypita, 2019-02-10:
I have done some test; it i difficult to arrange scale to not disturb the whole graphics, according to image quadrant and alignment. I have not found a nice look and feel that satisfies me

StuartB4, 2019-02-10:
Ok. Thanks for considering it.

arkypita, 2019-02-11:
[image]

Horizontal scale is ok. The major issue is on vertical axis.
If I want to put numbers on it, i should enlarge the space, or put them rotated.

StuartB4, 2019-02-11:
Oh yes I see. Would it be the same issue putting them down the right hand side?

arkypita, 2019-02-11:
I achieved a good result with the rotation of the text. For me it is satisfactory in all quadrants.
You will see it in the next version.

[image]

[image]

[image]

StuartB4, 2019-02-11:
Wow. That's excellent.
Looks really good, can't wait for the next version.
Thankyou.

arkypita, 2019-02-27:
done

closed 2019-02-27 by arkypita

StuartB4, 2019-02-27:
Thats great. 
Just tried it loading 1 image and Appending 3 others.
Got them all exactly where I wanted and the sizes I wanted.
Thanks very much for that, makes it a lot easier to position images.

Looks a bit untidy but it is only a test image.
[image]