arkypita/LaserGRBL #521: i can't  turn off my laser

xeriix, 2018-12-11:
i can't  turn off my laser because only works with M4: ON , M3:OFF and i can't edit laser off command in the program(LASERGRBL) . 
I have GRBL 1.1
Any help plz?

StuartB4, 2018-12-11:
M3 and M4 are both ON commands, M5 is OFF.

Because you have GRBL 1.1 you can go in to the GRBL configuration settings in LaserGRBL.
Start LaserGRBL, click on Connect so your machine is connected to a Com Port.
Click on the Grbl menu in LaserGrbl (top left) and click on Grbl Configuration.
Look down the list of settings for $32 and make sure it is set to 1.
Then click the Write button, then the Close button.
Like this:

[image]

xeriix, 2018-12-11:
The command to turn on work perfectly, the problem is with off, i have the same configuration you send ..

StuartB4, 2018-12-11:
Has it always done it, or has it just started.
What type is your laser and laser driver ?

xeriix, 2018-12-11:
Always , i have co2 laser with arduino uno

StuartB4, 2018-12-11:
Does it work ok with other different software ?

xeriix, 2018-12-11:
Yes , but work with M4:ON And M3:OFF

brakthehun, 2019-06-11:
Some CO2 power supplies need a negative trigger. You can invert the spindle in the config.h. you can also use an npn transistor to invert the signal. Do you have a pic of your power supply??

closed 2019-11-03 by arkypita