arkypita/LaserGRBL #522: Laser ON/ OFF Problems GRBL 1.1f Woodpecker board 3.x

JWChatham, 2018-12-11:
I have the grbl 1.1f woodpecker board and I am not getting the correct results from the software. It seems that sometimes the laser is opposite to what it should be. for example, when it should be on it is off, and vice versa.
What values should I use to turn on and off my laser that supports pwm signal.
I have tried 0-255 with no luck and 600-6000 with little luck. The laser seems to just stay on constantly in some configurations. Other configurations it is always off while running the program... not sure what is going on... any help would be appreciated.

Hardware:
 3180 CNC mod
Woodpecker 3.x board
Grbl 1.1 f

Software:
laserGRBL 3.0.4

JWChatham, 2018-12-12:
The problem was that M5 and M3 are inverted on my board.

closed 2018-12-12 by JWChatham

bmcminn, 2018-12-21:
How did you resolve this? I'm having an issue with a 3018 Pro Woodpecker 3.2 controller where my ACAN A12 laser module is always on and cannot turn it off...