arkypita/LaserGRBL #524: The contours of the picture when engraving

KameliAnna, 2018-12-12:
There is a black and white or grayscale picture (no difference). Black color is engraved on the power from settings. There is no contour at picture, but it seems to be engraved at maximum power. I used LineToLine with default settings. What could be wrong?
[image]
[image]

StuartB4, 2018-12-12:
Did you load a grayscale image in to LaserGRBL to start with?

StuartB4, 2018-12-12:
OK.
It would help if you could give us some info on your machine.
What type of Laser and laser driver board you have.
Also, what were the settings in here:

[image]

KameliAnna, 2018-12-13:
cnc shield v2.0 with arduino nano, 15W blue laser, LaserGRBL v3.0.4.
I made 2 simple tests with gradient picture:
[image]
First settings:
[image]
[image]
Second:
[image]
Result:
[image]
You can see black borders.

arkypita, 2018-12-13:
Use M4

arkypita, 2018-12-13:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

KameliAnna, 2018-12-13:
How can I change it? I have only M3 in combobox.

KameliAnna, 2018-12-13:
It helped! Thank you!

closed 2018-12-13 by KameliAnna