arkypita/LaserGRBL #529: Feature Request

JWChatham, 2018-12-17:
A.) Setting to Startup program with Laser GRBL on or off depending on users preference M5 or M3 depending on board.
B.) Setting Laser on to be M3 and laser off to be M5 for use with some of the Chinese boards that are backwards in sending the PWM signals.
After you press create have options under laser on: M3,M4, orM5 and options for laser off:M3,M4 or M5

This way you can configure your output gcode files to work correctly.

C.) An option similar to horizontal and vertical engraving but one that recognizes objects that are squared off and have sharp corners and only vertical and horizontal lines. IE so perfectly squared off objects will be printed correctly without rounded corners or edges chopped off etc...

JWChatham, 2018-12-18:
I found a way to invert the pwm signal with an Arduino using the map function. 
contact me for details. JWChatham@yahoo.com

arkypita, 2018-12-18:
Just to be absolutly clear:

LaserGRBL is a software for PC that run in windows and follow the specification of [GCODE](https://en.wikipedia.org/wiki/G-code) and its [GRBL](https://github.com/gnea/grbl) implementation where M3 is spindle/laser ON and M5 is spindle/laser OFF. S command is used to set spindle speed or laser power. 

The code generated by LaserGRBL is now and will be in the future full compatible with those specification so **i will not add any instruction to reverse M3 and M5 to make it "work correctly" because LaserGRBL is already working correctly.**

If your hardware does not follow grbl/gcode specification there is nothing to do in LaserGRBL and so there is nothing to write or propose here.

It is the responsibility of the GRBL firmware to translate the M3 / M5 commands into the correct electrical impulses. What you can do is configure your firmware or modify your hardware to meet these specifications. I have already give you links about how to do that here: https://github.com/arkypita/LaserGRBL/issues/525

closed 2018-12-18 by arkypita

JWChatham, 2018-12-18:
The solution is to use an Arduino to remap the pwm signal from 0-255 to 255-0.