arkypita/LaserGRBL #530: Too strong laser on S1

Scorotron, 2018-12-17:
Hi there. Is there any solution how to set weaker laser than S1? I added custom button M3 S1 for focus but it is too strong and burn the small point into wood or paper. S0.1 or S0,1 does not working, the same like S1. I have 2500 mW. Thanks

OldShark, 2018-12-17:
Hello. I did the same thing with mine. Added a button to focus the LD (M3 S1). my LD is 2.5 w. But I'm not using his driver.

arkypita, 2018-12-22:
Arduino can modulate PWM only in a range of 0-255 steps. This is an hardware limit so you can not have less then 1/255 of power.

However it sound very strange. I have a 2W blue laser and value less then S30 does not light up the laser at all.

Scorotron, 2018-12-22:
Thank so much for an answers. I have similar custom button, but M3 instead of M4. S1. But when I changed M4, it is the same. Focused S1 = hole to the paper or wood. But PWM should be working. I tried burn through more sheets of 80g paper and the more "S" parameter, the more pages burned through. On the pict very unfocused S1.

[image]

closed 2019-01-13 by arkypita