arkypita/LaserGRBL #531: No GRBL Configuration.

t518n47, 2018-12-18:
No GRBL Configuration.  When I click on Gbrl, Gbrl Configuration the Data is blank and I can't connect.  Moved the machine to another test computer and it works fine.  Configuration shows up and can connect.  Any ideas on how to fix this? 

I did note that the computer that it does not work on is a "64-bit Operating System".  And the computer that it works on is a "64-bit Operating System, x64 based processor"

Screen shot of the computer that works.  The computer with the problem does not have the top layer "Gbrl $$ configuration" 

Thanks.

[image]

arkypita, 2018-12-18:
> Gbrl Configuration the Data is blank and I can't connect

If you can't connect is normal that Gbrl Configuration the Data is blank. They are read from the machine: no connection = no read = no data.

If you don't connect maybe you have problem with your com port.
Did you have installed CH340 driver?

arkypita, 2018-12-18:
a copy of your session log can help
menu ? -> open session log

save the file and upload here

arkypita, 2018-12-18:
Usually the baud rate used by grbl v0.9 and v1.1 is 115200
In logfile I see a lot of try with others baudrate 38400, 19200, 9600 and only one try with 115200

You should use the correct baudrate if you want to connect. Do you know what is the correct baudrate for your setup? What is the version of grbl installed on your hardware?

t518n47, 2018-12-18:
No, I don't know the correct rate baud rate.  

Running grbl 1.1e

On two of my programs I know I connect successfully at 115200.

t518n47, 2018-12-18:
On one of the computers I can connect successfully at 115200.  I know because the controls in the lower left are accessible and I can move the laser.

Unfortunately I can't use that computer for long run.

arkypita, 2018-12-18:
115200 is fine for grbl 1.1e, do not use other values.

Configure like this:

[image]

t518n47, 2018-12-18:
Eureka!  Success.  

I really appreciate your assistance and how fast you diagnosed the issue.

arkypita, 2018-12-18:
Your are experiencing what is described here: http://lasergrbl.com/en/faq/

Maybe your hardware use a poor quality chinese arduino clone or electrical noise coming from stepper motor disturbing serial communication. Consider buy an original aduino or at least use shortest and shielded usb cable

t518n47, 2018-12-18:
Will change the cable.  Board is EleksMaker Mana SE v3.2.

simonez24, 2018-12-29:
Hi. I have same problem but my laser engraver is a very cheap version. I have kkmoon 3000mw and I don't know baud rate, I tested everything on your v3.0.6 but don't work.

simonez24, 2018-12-29:
Mi sono accorto solo ora che lei è Italiano e siccome anche il mio linglese è pessimo come lo è l'esperienza che ho in questo campo le espongo il quesito in madrelingua se non è un problema. Ho aquistato questo laser da 3w economico per produrre le bomboniere per il mio matrimonio, ora la cosa mi sta appassionando e mi ha spinto a cercare prima un nuovo software per vedere se riesco a lavorarci e successivamente un nuovo laser, ma sono fermo gia al primo approccio perchè non riesco a connettere l hardware al software. I driver sono installati e con il software originale anche se patetico ma funziona. Con la sua versione 3.0.6 ho provato a cambiare tutti i baud ma non si collega con nessuno e la com la sceglie in automatico tra 3 e 4 a seconda di dove metto la USB. Qualche suggerimento?