arkypita/LaserGRBL #532: Settings for 15 watt laser?How to lower power output?

JWChatham, 2018-12-18:
I have a really powerful laser. 15 watt.
Everytime I try to engrave something, I just cut through the wood.
How do I turn down the power? yes, I have pwm, currently set to 0=off 255=max.
Where do I turn down the power so I can engrave instead of cutting deep or through the wood?
Screenshots would be great.
Thanks.

StuartB4, 2018-12-18:
This should be set to M4.
You could also try changing the Engraving Speed just above it.

[image]

JWChatham, 2018-12-18:
will changing the Laser ON value  or the Laser OFF value reduce the power in any way?

JWChatham, 2018-12-18:
also can I get that test page you have there in the background?

StuartB4, 2018-12-19:
> will changing the Laser ON value or the Laser OFF value reduce the power in any way?

Laser On can only be M3 or M4, laser OFF can only be M5, so you can't change the values.
But thats what the S setting does, increase or decrease the laser power.
But that is calculated when you prepare an image and then generate the GCode.

Here's the test pattern you asked for.
[Test Pattern.zip](https://github.com/arkypita/LaserGRBL/files/2693039/Test.Pattern.zip)

StuartB4, 2018-12-20:
Sorry I don't understand, when do you need less power.

Usually in LaserGRBL you load an image, select the Conversion method, either Line To Line Tracing, 
1Bit BW Dithering, Vectorize or Centerline. 
Select the Direction and Quality, and adjust the Brightness, Contrast and White Clip if needed.
Then you click Next and it goes to the Target Image settings.
Set the Engraving Speed mm's per minute.
Set the Laser ON and Laser OFF. (M4, M5)
Set the S-Min Power to 0 and S-Max power to 255.
Select the size you want W and H and click Create.
LaserGRBL does the rest. Calaulating the GCode including the S power level setting
for that image based on your settings,

If you save the code with 'Save Program' on the File menu, then you can view it
or edit it if you want to.

This is part of one I saved, you can see the S values that correspond to the various
power levels for the laser.

G0 X0 Y20.588 S0
G1 X0.056 S234
X0.113 S196
X0.169 S165
X0.225 S217
X0.281 S245
X0.338 S173
X0.394 S197
X0.45 S226
X0.506 S157
X0.563 S148
X0.619 S167
X0.675 S197
X0.731 S255
X0.788 S249
X0.844 S243
X0.9 S241

arkypita, 2019-01-04:
> will changing the Laser ON value or the Laser OFF value reduce the power in any way?

If you configure $30=255, and in raster import you set S-MAX to 128 LaserGRBL will produce a range of S codes from 0 to 128, so you will use half power.

However I suggest you to use full power scale (0-255) and go 2 x faster

closed 2019-01-13 by arkypita