arkypita/LaserGRBL #538: Homing button.

RDS356, 2018-12-27:
Hi.
Why sometimes homing button doesn't appear ?

[image]

arkypita, 2018-12-27:
Homing appear only if you enable homing in $$ configuration.

If you don't have hardware with limit switch and homing enabled the homing button haw totally NO sense

RDS356, 2018-12-28:
Hi.
Yes, I agree with all that you told.
I am using soft limit. Look my $$.

[image]

RDS356, 2018-12-31:
Hi..
I have a commercial project that I use grbl 0.9i with Bcnc. There is more than 50 units sold around the Brazil.I made some changes in my grbl and change the version to 1.1 is not safety to me.
I'm working to change Bcnc to Grbl laser, but these littles problems still delayng this.
My intencion is, for each unit sold with Laser Grbl , I will make a donation of USD 20 to Laser Grbl Project.(Yes, I made donation to Bcnc and Grbl projects).
Regarding the low memory using grbl 1.1 with arduino uno, Did you have any problem ?
Happy new year.
Regards,
Reinaldo de Souza - from Brazil.

You can see my project here, unfortunately Brazilian portugese only;

https://produto.mercadolivre.com.br/MLB-842256838-plotter-de-caneta-rds-ed-tipo-axidraw-completa-_JM

StuartB4, 2018-12-31:
I have not had any problems with GRBL 1.1f on the Arduino UNO.
I know the GRBL uses all the space on the UNO's chip but it still seems to work fine.
The ATMega has 256K and the UNO only has 32k.
You can use an ATMega instead of the UNO with this version of GRBL:
https://github.com/gnea/grbl-Mega

arkypita, 2019-01-02:
> Hi.
> Yes, I agree with all that you told.
> I am using soft limit. Look my $$.

The condition for LaserGRBL to show-up homing button is to have $22=1
LaserGRBL detect $22 value when you configure $$ values using the configuration dialog, not if you change it typing $n=val

[image]

arkypita, 2019-01-02:
> Regarding the low memory using grbl 1.1 with arduino uno, Did you have any problem?

[As documented](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#i-get-this-low-memory-available-stability-problems-may-occur-warning-after-i-compile-grbl-is-there-a-problem) by grbl team there are no issue caused by the "low memory" warning of the compiler.

RDS356, 2019-01-02:
> > Hi.
> > Yes, I agree with all that you told.
> > I am using soft limit. Look my $$.
> 
> The condition for LaserGRBL to show-up homing button is to have $22=1
> LaserGRBL detect $22 value when you configure $$ values using the configuration dialog, not if you change it typing $n=val
> 
> [image]

Ok, now I Understood.
I did the configuration using other program, not laser grbl.
Thank you for explanation.

closed 2019-01-04 by arkypita

tus17, 2019-01-17:
Hello, 

i have a question about the home button. The homing button ist available, i have enabled the $22 in the configuration. BUT, when i click on the button, nothing happens. Can you help me?
i use the latest version of lasergrbl.

[image]

Kind regards,
Dimitri