arkypita/LaserGRBL #540: Laser logic backward?

ShaneBoss, 2018-12-31:
So i just received my 3018 cnc with a woodpecker 3.2 board grbl 1.1f and a 2.5w 3pin  laser.   
   The issue is when i set laser to low power i.e. m4s1  or m5 its at max power.   However if i set power to m4s255 it turns off and any high value like m4 s250 sets the power low.   [Like m4 s255 is less power than s240]
  Ive checked the pwm pin on the board and at m5 its showing 2.5v and with m4 s255 its showing 12v.
Is there an issue with my mainboard? Or is this a laser problem?

RDS356, 2018-12-31:
Well, where until I know, Sxxx only works with M3.

StuartB4, 2018-12-31:
Check your config.h file in "C:\Users\"Your User Name"\Documents\Arduino\libraries\grbl

 in config.h line206

> // Inverts the spindle enable pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful // for some pre-built electronic boards. // NOTE: If VARIABLE_SPINDLE is enabled(default), this option has no effect as the PWM output and // spindle enable are combined to one pin. If you need both this option and spindle speed PWM, // uncomment the config option USE_SPINDLE_DIR_AS_ENABLE_PIN below. #define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable.

If you change anything in there you will need to save the changes and then recompile GRBL and reupload to the Arduino.

ShaneBoss, 2019-01-01:
So I made the 2 changes in the above comment on my config file but I cant get it to upload so I saved the .hex that gets spit out in the temp folder and uploaded via xloader but it still does the same thing.  I also ordered another main board and laser both function the exact same 12v on vcc, 3ish v on pwm at idle and 12v at full power.

ShaneBoss, 2019-01-01:
Ok so I messed up earlier I was measuring voltage from vcc and pwm not ground but the logic is still backward and the above post didnt fix the problem but
 If I enable line 360 USE_SPINDLE_DIR_AS_ENABLE_PIN 
I can get it to work on pin d13 but pwm doesn't work only 5v or 0.
  Is there a way to get pwm on that pin?

brakthehun, 2019-06-11:
what version of GRBL are you using?? typically,, the laser TTL is connected to Pin 11 with GRBL 1.1 or pin 12 on GRBL 0.9. PIN 13 is spindle direction typically. PIN 11 has PWM capability.

closed 2019-11-03 by arkypita