arkypita/LaserGRBL #541: Raster result could be better - Tuning tipps?

maot1985, 2018-12-31:
Hi, 
I'm new to Github. I builded a simple CNC from wood and attached a laser with 2.5W to it.
Your programm works great, but I have question to my result shown below.

It seems, that the laser engraves deeper at the beginning and the end of a "raster line", where the laser changes the direction. It would be better, if the software lets the laser drive out of the image with the laser off. When the laser breaks and accelerates the laser is off so that there is no deeper burning and the result would be better. Is there such a function in LaserGRBL?

[image]

Kind regards,
Mathias Ott

jsp-email, 2018-12-31:
Gibt es eine Funktion wo erst die größe abgefahren wird  und dan erst geläsert wird

StuartB4, 2018-12-31:
Hi Mathias.
Which version of GRBL do you have ?
If it's version 1.1, do you have Laser Mode enabled ?
It's $32 and should be set to 1 in GRBL Configuration Settings on the Grbl menu at the top left of LaserGRBL.
Also if you are using PWM then set $30 to 255 and $31 to 0.
If you are using M3 to turn laser on try M4 instead.

Scorotron, 2019-01-01:
You have to set $32 to 1, then you can choose and use M3 and M4 as well. M3 has the same value independed on speed. M4 can change power in acceleration. You can set accelerations ($120-121) to the same speed as burning speed and then use M3. Bud be carefull ;-)

RunnyBore604, 2019-01-02:
@maot1985, you can try another software: https://yadi.sk/d/1WtJudo23ZugjZ
image2gcode adds accel/dexel distanxes, so you never gets overburned edges of image.

arkypita, 2019-01-02:
To chose M4 you should enable Laser Mode in GRBL configuration.

Do that using configuration window from "Grbl Menu"

[image]

LaserGRBL now know your configuration and will enable M4 in raster import window

maot1985, 2019-01-02:
I mentioned, that I enabled the lasermode, so here is my GRBL-Config for prove.
Again, I can not choose the M4 command. Is there any other setting wrong in my config?

$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=2 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=400.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=5.000 (Homing switch pull-off distance)
$30=255 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=400.000 (X-axis travel resolution)
$101=400.000 (Y-axis travel resolution)
$102=1600.000 (Z-axis travel resolution)
$110=3000.000 (X-axis maximum rate)
$111=3000.000 (Y-axis maximum rate)
$112=300.000 (Z-axis maximum rate)
$120=50.000 (X-axis acceleration)
$121=50.000 (Y-axis acceleration)
$122=5.000 (Z-axis acceleration)
$130=260.000 (X-axis maximum travel)
$131=260.000 (Y-axis maximum travel)
$132=100.000 (Z-axis maximum travel)

arkypita, 2019-01-02:
The solution is:

> Do that using configuration window from "Grbl Menu"

arkypita, 2019-01-02:
LaserGRBL can only know the board settings when configured through this window

maot1985, 2019-01-02:
Thank you very much arkypita for your help, now everything makes sense. Your software is great!

closed 2019-01-04 by arkypita