arkypita/LaserGRBL #543: Communication but no control

jringling, 2019-01-02:
I have a cheap machine with a bachinmaker board loaded with grbl 1.1z. The software appear to connect. I can read parameters with $$, but there is no movement with the buttons or sending gcode. What do I check next?

lfleck1, 2019-01-07:
I also have a Bachin machine (the 4050).  I was able to create a laser etching from a pic, but like you, was not able to use the manual arrow x/y axis buttons.  The etching came out great BTW.  I love the efficiency features of this software.

lfleck1, 2019-01-07:
Ok, I think I have a solution:  Apparently 1.1z may be a custom version(?).  I went to the grbl repository, downloaded to my PC, added it to my Arduino IDE as a Library, and then uploaded it to my Bachin-supplied Arduino Nano board.  the latest official grbl version appears to be 1.1g.  This seems to have fixed the issue.  NOTE:  I have NOT tested much within BachinDraw or Engraver Master to determine if there are any side effects of going to this version.

StuartB4, 2019-01-07:
I also think 1.1z is a custom version to suit the supplier of that machine.
I've seen it around the internet but never on the official GRBL sites.

Gabezoid, 2019-03-02:
Hi Ifleck, I think I have the same issue as your!
How do I add the grbl 1.1g to my Bachin board?
Sorry I'm not so into Arduino :(

brakthehun, 2019-06-11:
> Hi Ifleck, I think I have the same issue as your!
> How do I add the grbl 1.1g to my Bachin board?
> Sorry I'm not so into Arduino :(

Is it the same machine??
Most people use Arduino to upload new firmware - https://www.arduino.cc/en/Main/Software
then they download the latest GRBL firmware - https://github.com/gnea/grbl/wiki/Compiling-Grbl
and then they configure it - https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

closed 2019-11-03 by arkypita