arkypita/LaserGRBL #545: Bigtreetech cnc controller

garthanstey, 2019-01-02:
I have just received this Stepper driver but without any paperwork. The nano provided does not connect to my system but if I replace it with the nano from my old board it registers and allows lasergrbl to function but the motors dont respond. I've searched everywhere for a guide but cant find anything. I'm sure somebody must have one of these and can help. I just wanted to give myself a Z axis.

[image]

StuartB4, 2019-01-02:
Conflicting info on this board. I found 2 sellers with pics showing the stepper driver board facing opposite ways. Maybe you have them the wrong way.
[image]
[image]

garthanstey, 2019-01-02:
Thanks for that. What do I do with the 3 jumpers under the A4988 and  how do i connect the limit switches and ttl

StuartB4, 2019-01-03:
The 3 jumpers are for microstepping, but without instructions it's hard to say which jumpers should be on which pins.

StuartB4, 2019-01-03:
This may help with the microstepping.
https://www.pololu.com/product/1182

garthanstey, 2019-01-03:
The jumpers are marked M1,M2,M3. which fits with your comments,  so I will jumper them all as I don't need that much precision
I reversed the a4988s and the steppers now sing. Sadly they don't move at all just thump when you jog them.. The faster you jog them, the less they thump .I guess the board is shot. One of the power transistors did get extremely hot. Confirmed the limit right side is grounded which is a different ground to the drivers. Cannot find any connection  between Nano D11 and the outside wild so will have to rig something. Thanks again for your knowledge.

arkypita, 2019-01-04:
> I reversed the a4988s and the steppers now sing. Sadly they don't move at all just thump when you jog them.. The faster you jog them, the less they thump .I guess the board is shot. One of the power transistors did get extremely hot. 

Maybe you fry them, maybe not.

Driver should be regulated to drain the correct current for the specific stepper motor, however stepper doesn't move well. Read document and see video: https://www.pololu.com/product/1182

Driver are very fragile and sensitive device, if driver connected or removed while powered, if motor removed/connected while powered etc they will die.

garthanstey, 2019-01-14:
Ok. I now have a replacement board and the steppers are working. All micro step jumpers needed shorting and seems ok. But I can't get the laser to power up. The fan and laser don't get any power so I guess there is a jumper somewhere. I have found "s-en", "dir" and "c-en" in a group amongst the limiters. Any ideas what they are?
The board looks promising with opto-iso limiters and z axis. I'd love to replace my old board with this but need more info. Would appreciate any help

StuartB4, 2019-01-14:
I'm guessing here but I think  "s-en", "dir" and "c-en" 
are Spindle Enable, Spindle Direction and Coolant Enable.
For use if you wanted to use it as a proper CNC machine with a Spindle Motor instead of a Laser.
Still can't find any proper instructions anywhere for the board, which is a annoying really.

garthanstey, 2019-01-14:
Thanks for that. It makes sense. Still can't find how to trigger the laser or find any information about the board..

garthanstey, 2019-01-15:
Okay. I'm ready to scrap this board. I know it's got a model number LBA-B-V4 and is distributed by Bigtreetech.  But they didn't understand my request for more specific details. Is it possible that the firmware does a lot of the work and it has a dedicated version of grbl?

Rick23n, 2019-02-18:
I bought the same board that you did from Banggood but I had to return it as it was faulty and I suspect yours is also. I then bought a new board from Sainsmart (3 axis GRBL USB Driver Controller Board at this link (https://picclick.ca/GRBL-Laser-controller-board-CNC-USB-3-Axis-312323208146.html), which finally works a treat.See pic 
[image]
[image]
[image]
: I did not like the software that was supplied with it so to use Laser grbl the board has to be flashed with grbl 1.1 ( I think the one supplied had the previous version 0.9 which did not work) I have successfully added a z axis to my laser which which is controlled by a new button in Laser grbl. I am in the process of converting it to a cnc router for deep engraving of 5mm acrylic. I am adding a 48v 10a power supply for this.

garthanstey, 2019-02-20:
Thank you for the reply and photos. Very professional looking rig. I assumed that the board would function with a nano grbl1.1 system. But only the steppers ever worked. I guess it's a dud as you suggest. Your board seems to have nano integrated. What to do with all my old nano boards? My target was to end with a laser/router rig but I'm running out of time and my daughter wants to take it over as a plotter. She's nervous of that "sparkly laser thing" and the router noise and mess, and would prefer a plotter. I suspect i can create a servo Z and I have seen a pen lift with solenoid and powered by M3/M5. Thanks again for your response.

garthanstey, 2019-02-20:
Somehow managed to close the issue.

PierricG, 2019-03-13:
Hello,

A friend came at home with the same board and same laser issue. There is an issue with the board design which is quite easily fixable with an appropriated soldering iron. 
The laser EL817 photocoupler transistor side is not wired correctly. It's the one between R4 and C7.
If you look at the PCB traces, you'll quickly notice something is wrong.
Here's the fix:
- First disconnect any power supply from the board.
- unsolder R4 and keep it, we'll reuse it.
- with an x-acto knife (or similar), cut the trace that starts from the right side of R4 footprint and ends to the EL817 top lelf pin.
- solder a wire from a ground to the EL817 top lelf pin. Beware of galvanic isolation, the ground must come from the left part of the board, for instance D4 central tab (7805 voltage regulator) or C4/C5 capacitors.
- Now we need to power the photocoupler transistor side, we reuse the R4 resistor. You can use a standard 1kOhm resistor if you're not comfortable with surface mounted components. One side of the resistor must be connected to a 5V supply. I've taken mine from the left side of R5, located below R4.
- The other side must be connected to both the former left plot of R4 footprint (or right side of R3) and the bottom left pin of EL817.

Before testing with a laser, you can test with a fan. We noticed the laser command was inverted, but our board had the Q7 transistor on it, your doesn't have it (hopefully they did change the PCB accordingly).

Good luck.

Rick23n, 2019-03-13:
Hi PierricG
Thanks for that. I will give it a try as I still have the old Board and would like to have a back-up to the replacement. Will post my findings when done.

Rick23n, 2019-03-19:
Hi PierricG
Please can you confirm it is a 1k resistor not 10k as the others are?  And does it connect to left of R5 to left of R4 THEN to bottom left of EL817 OR from left of R5 to left of R4 AND to the left of EL817 at the same time? In other words is the resistor connected to EL817?
I hope you can help
Thanks 
Rick

BennettRockMike, 2019-04-19:
> So which one is right. I've taken it apart now! Have i damaged the circuit board? A power transistor got seriously hot.

So which way is the right way up? Dial top or Bottom?

closed 2019-06-23 by garthanstey

iula185, 2020-03-21:
Top picture is the right one...I think.. 
Measuring the connector for the motor with the pins on the driver board and GND pin with board GND this was the result. 
However I can't find anywhere the firmware for UNO, I wasn't able to turn any motor with this one and also teh 5V regulator is heating up like crazy. 
does anybody know where can I find the firmware?

christinoccs, 2022-04-22:
Hi, I have a HUGE problem with my laser engraver, I´m using arduino UNO or nano, in both product X axe only moves when the laser signal is sent, why just doesn´t move from side to side to the end and back, it makes me reduce the speed because of the vibrations generated