arkypita/LaserGRBL #548: Laser problem SW-LD73 V1.1

ofiryyy, 2019-01-05:
I bought a laser FROM EBAY in this link
 https://www.ebay.com/itm/450nm-15W-Laser-Module-W-Heatsink-Fan-Support-TTL-PWM-for-DIY-Laser-Engraver-J/222755022085?epid=2228220880&hash=item33dd3be505:g:gJgAAOSw0HFbbrsG:rk:1:pf:0&autorefresh=true

 And I bought the device + control card and engines step through another familiar ..
The problem is that the laser does not change its power. It engraves the image on the high volume and does not stop until you press the button on the card  SW-LD73 V1.1
[image]
[image]

[image]
Can this card work with this card?

StuartB4, 2019-01-05:
Laser power is controlled by the PWM connection which comes from the Arduino, or whatever board you use for GRBL.
If you dont have any PWM/TTL connection the laser will be on full power all the time.

ofiryyy, 2019-01-05:
My question is whether this card can work with this laser?
  And if so, how do I connect it?
  I'm really new at it.
How do I arrange the laser to work with PWM / TLL?

StuartB4, 2019-01-05:
Never seen one of them boards before.
Do you have any more info for it. 
It looks like It's an all in one Motor driver Laser driver and Arduino.

[image]

ofiryyy, 2019-01-06:
[image]

closed 2019-01-06 by ofiryyy

ofiryyy, 2019-01-07:
Do you have any idea with which card I can work?
To adjust my laser?

StuartB4, 2019-01-07:
The blue board should control the laser power by connecting one of the PWM connectors on the right to the Arduino's Pin 11.
It needs a voltage to run the board, fan and the laser. 

I can't be 100% sure because I can't find one of them boards that has the wiring diagram.
But I think it's like this.
12 volt in to the + and Gnd - and the PWM signal from Arduino Pin 11 in to the PWM/TTL pin next to them. The Gnd - should also be connected to the Gnd on the Arduino.

[image]

ofiryyy, 2019-01-07:
Can you recommend a card from eBay to the blue card?
Drive card for step

ofiryyy, 2019-01-15:
Can you recommend a high quality and powerful laser including a compatible electronic card?

StuartB4, 2019-01-15:
Sorry, not personally. I make my own with a laser diode, housing and separate PWM/TTL driver board.

[image]

ofiryyy, 2019-01-15:
Do you have a link to all this?

StuartB4, 2019-01-15:
I could give you links but there are hundreds of different types and prices.
Just go to Ebay and search for "laser diodes", "laser driver boards pwm ttl" and "laser diode housing"

The laser diodes I use are 5.6 mm so you should get a housing module to take that size if you use the same size diode.  Also, a lot of the housings have a plastic / acrylic lens, you should change it for a glass one, or buy a housing with a glass lens, and make sure it has the focus ring at the end like in the picture above. 
Some of them don't, then you need a Z Axis on you machine to raise and lower the laser to focus.

The driver board I use like the one in the picture can only work up to a 3 watt laser diode, so if you want a more powerful laser you would need a different driver board.

You will also need to solder the wires on to the laser diode yourself.
Just make sure you connect the pins together on the diode while soldering or handling, then remove the connection afterwards.
Use a soldering iron with an Earth/Gnd connection to minimise the risk of electrostatic damage to the diode.

I usually put the diode in the housing first, then solder the wires, that way the housing takes away some of the heat from the diode, and it's easier to solder.

ofiryyy, 2019-01-16:
I do not really understand that, right now I have the device as the LINK with 2 step motors and the green card.
I want to use the device as in the LINK and buy a strong and quality laser and suitable command cards and work with it ..
The aim is to etch on a quality wood, and cut a 3 mm wood and engraving on iron (if possible)
Because you understand that I wanted to link to something high quality.
https://he.aliexpress.com/item/DIY-500-mw-2500-mw-17/32867599520.html?spm=a2g0s.9042311.0.0.122e4c4dRuDnrz

StuartB4, 2019-01-16:
Because you already have the 15 Watt laser with the driver, you could do it this way.
An Arduino UNO (genuine for less problems with COM ports, also get the one with the 28 pin dil chip, it's replaceable if you damage it and cheaper than having to buy another Arduino).

A Protoneer version 3.51 Arduino CNC Shield. This version is fully compatible with GRBL 1.1.
You can only buy genuine versions of this.

2 Pololu Stepper Motor Drivers, either DRV8825 or A4988.

These are all available on Ebay.

[image]

ofiryyy, 2019-01-17:
The problem was that I returned the laser and the card to the seller
And now I'm looking for something quality and fit and I will not have any problems

StuartB4, 2019-01-17:
Oh ok.

ofiryyy, 2019-01-17:
I would be happy if you would advise me ..
You sure understand more than I do

ofiryyy, 2019-01-20:
?can you help me ?

StuartB4, 2019-01-20:
I would but I'm not going to say what you should buy if I have'nt tried it for myself.
As I've always made my own lasers and used various driver boards and motor drivers, I don't have any 
experience with other stuff. I've only used lasers up to 2 Watts so have no idea of higher powered ones.
You can buy everything you need in a kit but it usually includes the stepper motors and I assume you already have the actual machine with the motors on it.

You could look at this board, which should do everything you want.
But you will still need a laser  (PWM/TTL compatible) with driver board for it
https://www.ebay.com/itm/GRBL-Laser-controller-board-CNC-USB-3-Axis-Stepper-Motor-Driver-Controller-Board/323582293873?hash=item4b5701f771

Also read this post From Rick23n about this board. Ignore the first part, scroll down until you see the picture of the board.
https://github.com/arkypita/LaserGRBL/issues/436

josestrella, 2019-02-22:
[image]
[image]
Good afternoon, I bought this laser module and I have this arduino card. I am new in this world and I would like to know how can I put it to work?

Albatros64, 2020-08-13:
Hello! Since you have my own drive driving the laser, take a look here too! I'm having a lot of trouble getting it to work properly on my Marlin. I hope to be of some help.
https://madhacker.org/sw-ld73-laser-driver-modification/