arkypita/LaserGRBL #550: SVG File not interpreted correctly

mschaeck, 2019-01-05:
Hello,

wy is ist SVG shown diffentently than Inkscape?

[svgtest.zip](https://github.com/arkypita/LaserGRBL/files/2729788/svgtest.zip)

StuartB4, 2019-01-05:
Did you select all 3 objects and click the Object menu and Group them before saving.

mschaeck, 2019-01-05:
Also if I do it, it is not working.

mschaeck, 2019-01-05:
It seems to be a problem because of the program I creade the svg file. I I use Inkscape everything works fine.

But in Inkscape if I create a text, I could open the file correct with LaserGRBL. What do I have to to if I want to use text in inkscape to laser it.
[texttest.zip](https://github.com/arkypita/LaserGRBL/files/2729991/texttest.zip)

StuartB4, 2019-01-06:
Try the J Tech GCode generator extension.
Available here and where to put in in Inkscape and how to use it.
https://jtechphotonics.com/?page_id=1980

Select Object to path.

[image]

Select Extensions, Generate Gcode, J Tech Photonics Laser Tool.

[image]

That will create a GCode file of your text that you can load in to LaserGRBL using the File menu.

m3vuv, 2019-05-16:
seems the extension isnt there anymore on the website mentioned as far as i can see.seems a bum link.

StuartB4, 2019-05-16:
Link works fine for me.
Click this link and then check your downloads folder.
http://www.jtechphotonics.com/Downloads/Inkscape/JTP_Laser_Tool_V2_2%20-%20inkscape%209_2%20version.zip

[image]

m3vuv, 2019-05-16:
sorry my mistake,your fault lol,i thought it was an extension just for txt,sorry.

arkypita, 2019-05-18:
i suggest to transform all object, including text, to path

brakthehun, 2019-06-11:
What program are you using to generate the original SVG??

gisep, 2019-06-12:
I suggest the "Hershey Text" extension to laser text as one-line (and not 2 borders):
https://www.evilmadscientist.com/2011/hershey-text-an-inkscape-extension-for-engraving-fonts/

Same the process to transform to path before converting to G-code with "J Tech" extension

closed 2019-11-03 by arkypita