arkypita/LaserGRBL #553: consulta

naitsirc05, 2019-01-06:
hola, ¿como puedo hacer una vista previa del grabado láser en la pieza antes de grabar 
usando lasergrbl?

StuartB4, 2019-01-07:
Translation from English 1. Google Translator.

Haga clic derecho en una parte vacía de la barra en la parte inferior.
Haga clic en Agregar botón personalizado.
[image]

Copie y pegue este texto en el cuadro Gcode.

M4 
S10
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

[image]

Añadir una punta de herramienta.
Clic en Guardar.
Cargue su archivo de imagen y prepárelo para grabar.
Haga clic en Conectar para asegurarse de que su máquina esté conectada al puerto COM.
Haga clic en su nuevo botón personalizado para encuadrar el tamaño de la imagen y la posición en la pieza de trabajo con láser de baja potencia.
Para que pueda ver dónde se quemará cuando haga clic en Iniciar.

Translation from English 2. Microsoft Translator.

Derecha haga clic en una parte vacía de la barra en la parte inferior.Haga clic en Añadir Custom Button.
Copy y pegar este texto en caja de Gcode.
M4 
S10
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

Añadir una punta de la herramienta. Haga clic en Save.Load el archivo de imagen y preparar para grabar.Haga clic en conectar para asegurarse de que su máquina está conectada al puerto com.Haga clic en el botón nuevo y será el marco para el tamaño de la imagen y posición en la pieza de trabajo con láser de baja potencia.Para que pueda ver donde se quemará cuando usted haga clic en Inicio.

Translated from English.

naitsirc05, 2019-01-08:
muchas gracias, eso me funciono super bien.  ¿
se puede añadir mas funciones?

StuartB4, 2019-01-08:
Este botón personalizado es el botón de 2 estados y encontrará el centro de la imagen con láser a baja potencia.
Presione una vez para encontrar el centro.
Presione nuevamente para regresar a X0, Y0 y apagar el láser.

[image]

Gcode 1. Panel izquierdo

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

Gcode 2. Panel derecho

$32=1
G0 X0 Y0
M5

closed 2019-01-09 by naitsirc05