arkypita/LaserGRBL #555: fotos no se graban bien

naitsirc05, 2019-01-08:
hola. no e podido grabar bien una foto ya que salen muy mal.
¿que estoy haciendo mal? 

[image]

esta una de las configuraciones que usé 
[image]

StuartB4, 2019-01-08:
Intenta cambiar M3 a M4
[image]

naitsirc05, 2019-01-08:
no me aparece la opcion de M4  solo aparece M3

StuartB4, 2019-01-08:
Qué versión de GRBL tienes?
[image]

naitsirc05, 2019-01-08:
[image]

StuartB4, 2019-01-08:
De acuerdo. Para obtener M4 en el cuadro desplegable debe actualizar GRBL a 1.1f
Luego, obtienes 3 ajustes adicionales para el modo láser en los ajustes de configuración.
$ 30
$ 31
$ 32
[image]

naitsirc05, 2019-01-08:
Cuando instalo el 1.1 la máquina pierde la velocidad que tiene con el actual.
Esa velocidad se puede aumentar?

StuartB4, 2019-01-08:
Puede cambiar la velocidad en la configuración de la imagen de destino "Velocidad de grabado".
Pero solo puede ir tan alto como el ajuste en $ 110 y $ 111.
[image]

[image]

closed 2019-01-09 by naitsirc05