arkypita/LaserGRBL #556: Laser flash on connect

jabbott52, 2019-01-11:
I just started using LaserGRBL again (v3.0.4) after some time away.  Have to use hard reset to prevent steppers from jittering.  At connect the laser flashes on for an instant.  Works fine otherwise.  Thanks.

StuartB4, 2019-01-11:
I found that if I have just the Soft reset option enabled in LaserGRBL settings and connect the USB lead to my PC first, then power on my engraver, then start LaserGRBL and click the Connect button it does'nt flash.

jabbott52, 2019-01-14:
My problem may be due to a faulty usb to serial chip.  The problem now occurs whenever I connect the usb cable regardless of whether lasergrbl is running.  Usually several flashes during usb connect and the steppers bump.  Using woodpecker grbl board with fw 1.1f.  I'll try replacing the CH340 first.  Thanks.

maillemaker, 2019-01-30:
I see a laser flash sometimes on turn-on and turn-off of my machine.  I have not figured out if it is because I do not have a software hooked up to it or it not being plugged into the machine or what.

What I now do is before I turn the machine on or off I engage the "lower power laser mode" physical button on my machine.  This way if the laser flashes it is at low power.

Yell0w, 2019-01-31:
I have the same issue on my Eleksmaker A3 laser, i dont think this is a software issue, but the friendly guys over at their discord channel have this suggestion: 
**ThothLoki** 01/13/2019:  the issue is due to the time between the computer connecting to the nano and the nano loading the GRBL firmware. the laser pin of the nano is floating at this point and not pulled low so the laser will come on. 
The fix for this is to add a resistor (not a cap) between the "S" pin of the laser and GND. 
I cannot suggest a resistor value as i have seen so many different people suggest different values that worked for them.
I tried anywhere between 0.5 ohms to 100k ohms on my machine and my laser always came on. 
I just leave my board powered down until the GRBL firmware connects. That is my fix

brakthehun, 2019-06-11:
Many lasers are defaulted to on when connected. Most controllers take a quarter second to boot. during the boot process,, the control pin is "open" which let's the laser control it's self and let's it be on.

One quick work around for this is to hook usb up first,, then turn the laser machine on.

jabbott52, 2019-07-02:
I was having multiple laser flashes, not just a single one at connect.  I tried replacing the CH340 chip but that didn't help.  What did fix the issue was removing the C6 capacitor from the woodpecker.  This capacitor (on my board) ties the DTR from the CH340 to the RST (reset) of the ATMEGA.  Apparently the CH340 was causing multiple resets of the ATMEGA.  Still don't know what changed from when I first got the woodpecker.  I was using a different computer so perhaps the USB on the new computer is wonky.  It occurred on both Windows and Linux.  Anyway,  this might help somebody having the same issue.

closed 2019-11-03 by arkypita