arkypita/LaserGRBL #557: New User - distortion on vector image

maillemaker, 2019-01-12:
Hi all,
I have recently acquired a used Eleksmaker A3 Pro.  The machine seems to be functioning correctly.  However, I am seeing some distortion when burning a vectorized version of an image.

This is what I get with dithering:
[image]

This is what I get with vectorizing:
[image]

I have noticed that the belts have different tension on one side of the unit vs. the other.  I will take a look at that in a bit.  I'm not sure how tight they are supposed to be (I have a post on the Eleksmaker forum also).

I have set the resolution on the dithered image to 10 lines/mm.  If I am understanding the UI correctly device supports up to 80.

Burn speed of the dither was 1500.  Burn speed of the vector was 500.

Any other ideas to check?

einencool, 2019-01-13:
To me it looks like the belts are too loose.
I started with mine a few days ago and at the beginning the belts were also to loose and then I changed the holders that are tightening the belt, and now everything is fine.

A had the same issue and today I made some tests and made a circle with 50mm diameter and tested it 20 times on the same position with not much Power, and all rings are on the same line.

Greets Chris

maillemaker, 2019-01-14:
I have tightened the belts.  I have also turned off bi-directional burn mode.  In pure vectorize mode you can clearly see the machine is having difficulty with concentric circles.  In vectorize+fill mode it is somewhat better, but I think the fill is just hiding the problem.  You can see even in the fill image that the cannon wheel rim is out of register.

Artwork (printed to 100 mm):
full resolution: https://i.imgur.com/4hotiJ1.gif
[image]

Vectorize only:
[image]

Vectorize with fill:
[image]

einencool, 2019-01-14:
Is this Problem only when you're working with the Code made by Lasergrbl?
When you get the same Results  with another Software, I think the Problem is not in the Software.

Sadly I'm not able to attach my .NC-File I made with another Software. When I try to attach it as a *.txt or something like that, I always get an error message...

einencool, 2019-01-14:
Right now I've burned your Picture in 80x80mm on a sheet of cardboard.
I think it turns out pretty well

[image]

maillemaker, 2019-01-17:
That sure looks good!

OK, I took a good look at this machine.
The first thing I noticed is some of the plastic spacers for the rollers were cracked.  So, I took some 1/4" x .014" wall brass tubing and cut a bunch of 10mm spacers:

[image]
[image]
[image]

The next thing I noticed is the machine was not bolted up square, so I fixed that:

[image]

I made up a test image of some concentric circles.  This printed out very well in Vectorize mode.
[image]

Then I tested out my "real" image in Vectorize mode, and it worked great also (see bottom image in this photo):
[image]

But when I try to print the same image using Vectorize with diagonal fill, it smears the image (see the top image in the above photo).

Here you can see another example:
[image]

The top image is Vectorize with Fill.  You can see that it smears the fill, but the vector borders burned perfectly.

The bottom image is when I tried Dithering.  It also distorts.

arkypita, 2019-01-17:
can you post your grbl configuration here?

maillemaker, 2019-01-17:
OK I will do so when I get home this evening.  Thanks.

einencool, 2019-01-17:
Well I think that your problem is still in the hardware.
Can you check the power of the drivers given to the stepper motors? Do you have 1 or 2 drivers for your Y-axis? If there are two, do they have the same voltage?

Second thing, and I guess it will be more problematic, how tight are your axis unpowered?
When you didn't have the belts attached to the pulleys are they really smooth and without the need to push?
Mine will drop when I lift the machine slightly without the belts.

Although on your picture with the vectorised image. The problem you had is still there...
Please check the hardware. In my opinion it's not a software problem.

arkypita, 2019-01-17:
While filling there is a lot of rapid movement used for skipping white space. Rapid movement are commanded via G0 and use to move at the highest speed available, using acceleration and deceleration ramp.

If your setup is not well fluid you could have inertia and loose step, resulting in a "compressed" distortion where white space are smaller then expected.

In vector mode all movement are done with G1-G2-G3 that are linear movement. Only placement between different path are done with G0.

This is why I ask you to put your config here.

Maybe your value of accell and max rate are too high

$110=500.000 | X Max rate, mm/min
-- | --
$111=500.000 | Y Max rate, mm/min
$112=500.000 | Z Max rate, mm/min
$120=10.000 | X Acceleration, mm/sec^2
$121=10.000 | Y Acceleration, mm/sec^2
$122=10.000 | Z Acceleration, mm/sec^2

arkypita, 2019-01-17:
My suggestion: use a simpler image to do your tests.

arkypita, 2019-01-17:
[image]

maillemaker, 2019-01-17:
Here is my current configuration:

$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=3 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=400.000 (Z-axis travel resolution)
$110=5000.000 (X-axis maximum rate)
$111=5000.000 (Y-axis maximum rate)
$112=5000.000 (Z-axis maximum rate)
$120=200.000 (X-axis acceleration)
$121=200.000 (Y-axis acceleration)
$122=200.000 (Z-axis acceleration)
$130=5000.000 (X-axis maximum travel)
$131=5000.000 (Y-axis maximum travel)
$132=5000.000 (Z-axis maximum travel)

maillemaker, 2019-01-17:
I will try your test image with my default values and then reduce the $120 amount by half and try again.

maillemaker, 2019-01-18:
I believe I have discovered the cause of the distortion.  I saw a YouTube video where a guy was using this engraver to make PCB boards and so needed microscopic accuracy.  He made special hold-fasts for the drive belt ends because the factory ones cause uneven tension as the trolleys move towards the belt ends.  This is because the factory belt keepers push the belt down into the 8020 grooves, so as the pulleys move close to the belt ends they have to pull the belt up out of the groove and this gets harder to do the closer you get to the keeper:

[image]

To test this, I first burned a test file after I homed the system to approximately dead-center of the rig.

[image]
[image]
[image]
[image]

This worked fine.

Next I burned a 50mm version of my logo I have been trying to print.
[image]
[image]

This also printed pretty well, though there is still some alignment issue.

Finally, I printed the 100mm version I had been trying to do:
[image]
[image]
[image]

This did pretty well.

Then I re-homed the system at the bottom-left corner, near the belt keeper, and tried the 100mm image again.  Sure enough, distortion like last night.

So, I think the problem is that the belt tension changes as you get near the belt ends, and this is throwing off the burn.  In the YouTube video I saw with the PCB guy, he made special 3D printed keepers that kept the belt at a uniform height in the 8020 channel throughout the range of the unit.

Still not sure why the vector burn worked fine while the fill burn did not, but anyway, I think it's pretty conclusive that the machine has a problem when you reach the limits of the chassis.  I'm going to try and rig new keepers for the belt like the PCB guy did.

maillemaker, 2019-01-18:
Here is the PCB fellow who added some 3D printed parts to alter the way the belts terminate and also to allow tension adjustment of the trolleys.

https://www.youtube.com/watch?v=ZzO9skEYqwY

Steve

maillemaker, 2019-01-18:
I believe this issue is mechanical in nature and resolved now (or at least known how to resolve) so I will close the issue.

einencool, 2019-01-18:
I think it would be great if you could post your final result after you made your optimization on your machine.

When you made your logo in the mittle of the machine there are absolutely the same problems as you mentioned before. But because it's a little bit smaller you say "it turns out pretty well". Well in truth it's really the same problem.

And the "new" belt holders will not make such a difference.

I told you to check how your units run on the rails. When they don't move play free AND without the need to push you should correct it.
When I built my machine I tested all of the 3 units and they slide absolutely smooth.
Wenn I flip one of the rails with one unit around about 20 degree it will start to run without I have to push it....

maillemaker, 2019-01-18:
The vector distortion is there, but is better after I rebuilt the machine.  I made sure to square the frame, and I tried to tune the position of the rollers so that the trolleys rolled somewhat easily but without any play.  There is still too much resistance, I think.  

It is _probably_ good enough for my purposes of woodburning designs, especially with fill.  But, I have also discovered 3D printable wheel holders that allow adjustment of the wheel tension so that the trolley can run freely but without any play on the 8020 rail.  I am going to download these and print them out.

What I don't understand is if the motor keeps track of steps, and the steps don't change (no teeth jumping on the belt), then I would not expect any distortions of the image due to machine friction; only the motors would have to work harder.  I don't understand what causes the distortion.

einencool, 2019-01-18:
Are your steppers loosing steps?
That's why I asked for the voltage of the stepper drivers.

Can you see on which axis the shape is not ok? X or y or both? Maybe you make a circle with a little spot in the center and then paint a circle with a compasses (right word??)

StuartB4, 2019-01-18:
Hi Maillemaker.
I noticed you are still using GRBL 0.9. Have you thought about upgrading it to 1.1.

maillemaker, 2019-01-22:
Wow I only downloaded it a few weeks ago.  I'll get the latest.

maillemaker, 2019-01-29:
Well, I still have the problem.  I thought it had gone away due to my test pieces.  I burned a final test piece at the target size of 120 mm, and then moved to my project piece of wood.  Distortion in the fill but not in the vector border:

[image]
[image]

So, I put down a piece of scrap wood and changed the X and Y acceleration to 50.  Distortion is worse.  It is burning near the center of the chassis.

[image]
[image]
[image]
[image]

I have 3D printed some bushings to replace my brass tubing.  But I don't think that is the issue.

This is out in my garage.  It gets colder at night.  Could temperature be a factor?

Also, I went and bought one of those shielded, ferrite core USB cables and used it in the above burns.

arkypita, 2019-01-29:
> This is out in my garage. It gets colder at night. Could temperature be a factor?

I do not think it can affect so much

> Also, I went and bought one of those shielded, ferrite core USB cables and used it in the above burns.

This is good. You can also try to set "quiet" as threading mode, to see if it was caused by communication issues. 

However since v2.8.40 engraving command are generated and sent by LaserGRBL using absolute coordinates, so if there are any serial communication error between PC and only single segments should be interested, while the whole drawing should be ok.

arkypita, 2019-01-29:
I am very convinced that this is a hardware problem. You must verify that your engines do not lose steps at max speed, that is the speed used for G0 rapid movement.

Try this procedure:

- Move laser head to bottom-right of your table (X0 Y0)
- Place one 1.5V AA battery (or anything that can stand vertically) at the point where the laser head is very near the head, better if touching the laser head.
- Set jogging interface to very slow speed (i.e. 100) and big step (i.e. 200)
- Press the diagonal up-right arrow to move to X200 Y200
- Place another one AA battery to mark down this position. You should place both batteries so that the laser does not drop them when it moves diagonally.
- Now set jogging to the higher speed (maybe 5000).
- Press diagonal bottom-left arrow. The laser head should return to X0, Y0 in the same initial position touching the battery (or the object you use as reference) but without hitting down them.
- Loop this diagonal movement more times, always check that reached position is good.

If diagonal is ok try singular X and Y movement in the same way.

arkypita, 2019-01-29:
[image]

Something like this

arkypita, 2019-01-29:
Something like this

[image]

maillemaker, 2019-01-29:
I have disassembled the unit again.  I discovered that one of the roller wheels, which contains 2 ball bearings, did not have the needed spacer between the two ball bearing inner races.  So when the axle bolt was tightened, it was squashing the two inner races towards each other, possibly causing the bearings not to spin.  I made a spacer and put it back together.

The main gantry (what I call Y axis), seems to move quietly and easily the whole length of the beams, but the gantry does not roll freely if you stop pushing.

The laser head (what I call X axis) is much stiffer.  When I just went to check on it, when I slid the laser head back and forth there were definite "bumps" at regular intervals.  I put a small pencil dot on one of the rollers, and it corresponds to one revolution of the roller.  I suspect a flat spot on a roller.  But, I loosened all up, and attempted to bias the rollers as far outbound as possible in their fixed holes, and now the bumps are gone.  I wonder if the rollers are plastically deforming after sitting to take on a bump from the rail.  I cannot feel any deformation son the rail or on the wheels with my fingers.

I will try the battery test shortly.

maillemaker, 2019-01-29:
OK, I set the speed to 78 (could not slide the slider to hit 100), and the step to 200.

My AA batteries were not tall enough to reach the laser head, but I think you are just looking for a reference spot, so I turned on the spotting laser power and marked the start position, and then marked the end position, and cycled back and forth.  It looks pretty repeatable.

https://youtu.be/zOeMpYWaC6c

I am now doing a test burn on my logo.  I uninstalled and re-installed lasergrbl.  But it seemed to remember my settings (50 acceleration).  So, I bumped the acceleration back up to 200 since that did not seem to have an effect anyway.

arkypita, 2019-01-29:
All the $ setting are stored into hardware.

maillemaker, 2019-01-29:
Test burn is already showing distortion.
https://i.imgur.com/1H7E8eW.jpg

maillemaker, 2019-01-29:
I have fired up the EleksCAM software (provided with the device) and it seems to be burning fine.
I am not familiar with this software so I am not sure I am exactly comparing apples to apples with the options, but I think I'm doing a vector-fill burn.

[image]

It is strange that the lasergrbl vector burn works fine while the fill is distorted.  If this were a mechanical issue I would expect distortion in both.

maillemaker, 2019-01-29:
Here is the output from the EleksCAM software using the PicCarve option.  Seeing the final output I do not believe it does a vector trace - it appears to be a straight fill.

[image]

But, fill is what seems to be giving LaserGRBL trouble.

I have installed LaserGRBL on another computer (so I can use this one for other things), and I am in the middle of a vector-only burn test with LaserGRBL.  Will post results shortly.

maillemaker, 2019-01-29:
> Hi Maillemaker.
> I noticed you are still using GRBL 0.9. Have you thought about upgrading it to 1.1.

I installed the latest LaserGRBL.  I do not know how to upgrade GRBL by itself.  How do you do that?  

Steve

maillemaker, 2019-01-29:
Did a vector-only burn from LaserGRBL.  There is still a tiny bit of distortion in the concentric circles, but this is far different from the fill distortion problem.
[image]

maillemaker, 2019-01-29:
On the new computer, by default it is bi-directional burning once again.  
Here is LaserGRBL, 120mm output, bi-di burning, vectorize with fill, horizontal fill:
[image]

StuartB4, 2019-01-30:
> I installed the latest LaserGRBL. I do not know how to upgrade GRBL by itself. How do you do that?

Depends which controller board/arduino you have.

maillemaker, 2019-01-30:
GAH!  

You guys are not going to believe this.

I believe I have identified some of the misalignment issues I had been seeing.  I don't think this accounts for the large "super smear" problem, but definitely the smaller ones.

The machine itself is "walking" around the work surface due to the movement of the gantry and laser head!

Last night, on a test burn that was showing some distortion, on a whim I decided to take a pencil and mark one of the polycarbonate feet.  I came back from time to time and re-marked the foot.  As you can see, the foot is moving on the work surface!

[image]

One of the reasons this was difficult to realize is that my test work surface is a rougher surface and less prone to "walking".  But when I moved the chassis to the work piece I am intending to burn for real, which is sanded nice and smooth, it now easily slides on the work surface.

This would also explain why the fill gets distorted, but the vector border burn does not.  The fill is done with many quick, jerky movements of the laser head.  But the border fill is burned with nice smooth, continuous movements.  So the fill is more likely to walk the machine around the work surface than the border tracing.

I'm going to get some double-sided tape tonight to put under the feet, which should stop the machine from moving on the work surface.  I will then re-run some tests.

Steve

StuartB4, 2019-01-30:
Just another thing to confuse you. Something to remember for the future.

maillemaker, 2019-01-31:
Did more testing last night with the feet held secure with double-sided tape on all 4 feet.
With vector border tracing only, I still get a little bit of non-concentricity at two points on the circles.   With vector border + fill, I got good results. 

[image]
[image]

I'm still not convinced that the "super smear" problems were caused by the machine walking.  But, I don't know what caused them otherwise.

Steve

arkypita, 2019-01-31:
[image]

Also tries to understand why the beginning and end of the circles do not coincide.
This should not happen.

Focus on drawing circles, at different speeds. See what happen

einencool, 2019-01-31:
When you look at the picture in the first post you already see those tiny gaps between the start and end point.
My opinion doesn’t change and I still believe that the issue is in the hardware...
Maybe the wheels are pushing to strong onto the rails or the belt tension is not set correct.

And you still have issues with the space between the lines in the lower left and the upper right direction. You can see it in EVERY picture you posted here. 

That is definitely not an issue from the software.
Have you measured the power for the stepper drivers?

maillemaker, 2019-02-01:
I don't know how to measure power for the stepper drivers.

I think the belt tension is fine.  But, I am going to replace the belt keepers with 3D printed ones that stop the belt from sloping downwards in the channel towards the ends.
I do think the wheel tension is too tight and it is causing flat spots to form on the wheels.  At least on the laser head.  After the unit has sat idle overnight, if you manually drag the head back and forth you can feel a "thunk" as it passes over the flat spots in the wheels.  I suspect they are gripping the rails to tightly and so plasticly deform when left to sit.   I am also 3D printing auto-tensioners to hold the bottom wheels in lightly but firmly in place.

But I really don't expect anything to get much better.  I suspect this is about as good as this machine can do.  But I will keep testing as I add the new parts.

einencool, 2019-02-01:
Have you checked the tension of the wheels without the belt?
On every axis it must be as loose as you can go But it shouldn’t be to loose...

maillemaker, 2019-02-04:
I'm working on that now.  I have downloaded new wheel holders that have built-in leaf springs that provide tension.  Right now the tension is maintained solely by the axle-to-axle distance, and it is tight.  The tensioners should fix that.

maillemaker, 2019-02-05:
I had to tweak the design of the roller tensioners as they did not exactly fit.  Plus I was able to simplify the design.  I have printed out and installed the tensioner set for the laser head carriage, and it now slides back and forth on the rail MUCH more freely than before.  Before, the wheels gripped the 8020 channel so tightly you could turn the rail nearly vertical before the carriage would start rolling.  Now you can only tip the rail about 15 degrees from horizontal before it starts rolling.

I am now printing out the tensioners for the side carriages.  Then I will print out the new belt keepers.  Then we'll see what kind of results I can get.

maillemaker, 2019-02-08:
Gah.  Design flaw in side tensioners.  Have to fix and re-print!

maillemaker, 2019-02-24:
Well, I'm not having any luck.  After installing my 3D printed self-tensioning wheel holders, the thing printed even worse.  So, I removed them all and put it back stock, and now it still cannot print concentric circles worth a damn.  I have tried really tight belt tension (which caused the motors to stall), and I tried loose tension.  I'm about to give up on this device.  Definitely not a turn-key solution.

maillemaker, 2019-05-13:
Started playing with this again tonight.  I have put the machine back to stock configuration with the roller wheel mounting, but I have put 3D printed belt keepers in place that have a protrusion that keeps the belt level in the track.  Also my wheel spacers are 3D printed as the eleksmaker spacers had broken.

[img]https://i.imgur.com/l0faHY5.jpg[/img]

You can see I'm still getting distortion on circular objects.

I have experimented with running the belts very tight - this causes the motors to stall sometimes and the unit to lose position.  I have experimented with them loose, and in between.  

I have played with the Settings, which I understand writes them to the machine.

How do I reset the machine back to factory defaults?

How do I reset LaserGRBL back to its factory defaults?

maillemaker, 2019-05-18:
OK, after much head scratching and fiddling, I have fixed my problem.

The problem was 2 broken connections on the left gantry motor.  This resulted in it not pulling its weight.  When I fixed the broken connections, the machine started operating perfectly.

broken leads:
[image]

Test after fixing:
[image]

closed 2019-05-18 by maillemaker

StuartB4, 2019-05-18:
Brilliant, after all the messing around and head scratching.
Happy you got it sorted.