arkypita/LaserGRBL #558: Question Hotkey and Two State Button

einencool, 2019-01-12:
Hello,
at first, thank you for your great software. I'm new to Laser and to LaserGRBL.
Now I'm configuring my Hotkeys like on my "normal CNC-machine"
You have the great Feature with the "Two State Buttons".

But I think I'm doing something wrong.
I use a Two State Button for centering the Laser in the middle of the picture.
When I press it once, The Laser goes to the center and turns on the Laser
Pressing it the second time, the Laser turns off and went back to Zero.

But when I'm doing it with STRG+1 it only makes the first step.

Is there anything I'm doing wrong.

The code is:
#1:
G1 X[left+width/2] Y[bottom+height/2] F1000
M3 S1

#2 
M5 S0
G1 X0 Y0

Thank you in advance.
Greets Chris

einencool, 2019-01-13:
Hello Stuart,
Yes like I said, when clicking  the "Custom Button no.1" everything works well.

But when I use the hot key it is originally set to "ctrl + 1" ( Sorry in german it is "Strg + 1")
It will only repeat the first step every time I press the key.

On my cnc mill I use a separate usb num pad to control the machine. And it would be great, when I can control the laser with just one key.

Greets Chris

StuartB4, 2019-01-13:
Ok, now I understand..
Does the same on my machine also.
I found a way by just adding another custom Single press button with the GCode 2 box settings from the 2 state button.
Then use  STRG+1 to center and STRG+2 to return.
On mine it's CTRL+6 and CTRL+7 because I have other buttons assigned to CTRL+1,2,3,4 and 5.

[image]

einencool, 2019-01-13:
Yes sure, I know that it would be possible with two buttons, but I thought it would be possible to make it only with one button.

Maybe it's a little bug in the software and maybe it can be fixed in the future.

Greets Chris

arkypita, 2019-01-13:
thanks @einencool 
i will fix as soon as possible

einencool, 2019-01-13:
Thank you for your continuous support

arkypita, 2019-01-13:
can you try and deep test new version https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.7

arkypita, 2019-01-13:
let me know

einencool, 2019-01-13:
Wow, you're lightning fast :-)

I tried it now only without burning, because I've got a copy of my Arduino next to me.
Everything seems to work like it should.
I tested it with some two state buttons and the normal ones.
Then I started engraving (without hardware) and everything seems to work fine.

Thank you for your extremly fast support