arkypita/LaserGRBL #559: Connecting fails but machine turns on and is uncontrollable

Aztaro, 2019-01-16:
I have a 3018 woodpecker CNC and the CNC part works fine with it's native program but when trying to use Laser GRBL with a 2500mw laser, the Y axis motor turns on and the laser turns on full power when trying to connect but after a few seconds the program disconnects and the Y axis motor and laser keep running [almost set the table on fire]
I have 0 idea what to do since it won't finish connecting, session log only has

1/16/2019 1:45:03 AM	Program     	------- LaserGRBL v3.0.4 START -------
1/16/2019 1:45:05 AM	SetStatus   	Machine status [Disconnected]
1/16/2019 1:45:25 AM	SetStatus   	Machine status [Connecting]
1/16/2019 1:45:25 AM	OpenCom     	Open COM4 @ 115200 baud 
1/16/2019 1:45:25 AM	SendImmediate	Send Immediate Command [0x18]
1/16/2019 1:45:35 AM	OpenCom     	Connection timeout!

Tried every "Baud" setting and same results, even with Laser GRBL program that came on the disk has same problem

arkypita, 2019-01-16:
Are you sure native program use grbl?
lasergrbl.com/faq

Aztaro, 2019-01-16:
It came with Laser GRBL 2.9 on the disk

arkypita, 2019-01-16:
did you have installed CH340 driver?
Right baud is 115200

Aztaro, 2019-01-16:
Windows 10 had a driver set up but used the program to uninstall it and reinstall it

arkypita, 2019-01-16:
There are many different software like LaserGRBL that can be used with grbl compatible hardware. You can find a list here: https://github.com/gnea/grbl/wiki/Using-Grbl

Does others software work or not? I suggest to test with Universal G-code Sender (UGS)

If not maybe you have received a broken board.

Aztaro, 2019-01-16:
I'll test it right now
Also how could the board be broken if when using Candle and a spindle it works fine?

arkypita, 2019-01-16:
So, with Candle it work. Right?
Should be a difference in how Candle and LaserGRBL manage connection initialization.

Try configure LaserGRBL like this:

[image]

Aztaro, 2019-01-16:
That solved EVERYTHING
Thank you very much, I'm running a test as we speak and it both connected and is following the program so far

arkypita, 2019-01-16:
This is a very safe configuration. You can configure streaming mode to buffered and threading to normal/fast

arkypita, 2019-01-16:
I guess it should work too

Aztaro, 2019-01-16:
If I could ask an off topic question real quick, is there a way to turn the laser on a "low power" mode so I can focus it without having to run it?

arkypita, 2019-01-16:
Yes, with custom button.
Do a search in closed and opened issue "share your custom button here"

Aztaro, 2019-01-16:
Thank you very much for your time, I hope you have a good 2019

closed 2019-01-16 by Aztaro